Klipper3d/klipper #3862: Unable to read tmc uart 'stepper_x' register IFCNT

hlgithub7430, 2021-01-30:
MKS_SGEN_L V2.0  Mainboard  | TMC Stepper Fysetc 2209. on all axis. 
Action already done:
Swapped 2209 between X and Y. 
Set UART_ADDRESS to 0 and 1;2;3 nothing.
Checked voltage to 2209.
Control jumper setting according MKS 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5897855/klippy.log)

boardthatpowder, 2021-01-31:
<microsteps> has moved from tmc* sections to the stepper_* sections as part of https://github.com/KevinOConnor/klipper/blob/master/docs/Rotation_Distance.md. This needs to be added to config_check.md as is not clear.

danobot, 2021-01-31:
@boardthatpowder has this change been officially released? Last tag on master is from October 2020. A bit confusing that the docs arent versioned along with Klipper releases.

boardthatpowder, 2021-01-31:
Officially released - no idea. I’m working from master and hit this same problem over the weekend when I pulled latest.

boardthatpowder, 2021-01-31:
Wait - my comments look like they’ve gone onto a different ticket (using phone), and I can’t find the original one now. My comment referred to someone questioning why they were getting an error regarding microsteps not defined after updating.

hlgithub7430, 2021-01-31:
Hi, maybe to use step_distance is also a problem, so I have to convert to Rotation_Distance. But I guess my error is more in communication between TMC and Klipper.

linux-paul, 2021-02-19:
I'm running a similar configuration with the same Board with
[tmc2209 stepper_x]
uart_pin: P1.1
is fine.
[printer.txt](https://github.com/KevinOConnor/klipper/files/6010721/printer.txt)

hlgithub7430, 2021-02-25:
Thanks very much. I will try your settings.

KevinOConnor, 2021-03-05:
Given our updated policy on github tickets (see https://www.klipper3d.org/Contact.html ) I am closing this ticket.

-Kevin

closed 2021-03-05 by KevinOConnor