Klipper3d/klipper #3864: MKS Robin E3 Problem with flashing firmware

Salkio, 2021-01-30:
Each time I try to make flash... I get this error

_dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

Failed to flash to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=1eaf:0003
  OR
  make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0_

I tried different bootloads i tried everything in my knowledge. 
Inside octoprint when i "Get Status"
I get:

_5 Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
Klipper state: Not ready_

[klippy.log](https://github.com/KevinOConnor/klipper/files/5898174/klippy.log)
[klippy2.log](https://github.com/KevinOConnor/klipper/files/5898178/klippy2.log)

Sineos, 2021-01-30:
https://github.com/KevinOConnor/klipper/blob/master/config/generic-mks-robin-e3.cfg

Quote:

```
# Note that the "make flash" command does not work with MKS Robin
# boards. After running "make", run the following command:
#   ./scripts/update_mks_robin.py out/klipper.bin out/Robin_e3.bin
# Copy the file out/Robin_e3.bin to an SD card and then restart the
# printer with that SD card.
```

Salkio, 2021-01-30:
> https://github.com/KevinOConnor/klipper/blob/master/config/generic-mks-robin-e3.cfg
>
> Quote:
>
> ```
> # Note that the "make flash" command does not work with MKS Robin
> # boards. After running "make", run [...]

Thank you so much ! I’m going home shortly and I’m going to try this. 
I did use that config you linked but am really confusing about the BLTouch and Filament sensor. But let me see if what you said works for me ! Have a great day further.

Salkio, 2021-01-30:
Do you know perhaps know what settings I should use in the make config ? Especially the boot loader.

Sineos, 2021-01-30:
Would you be so kind to consider reading the damned config file?

Quote:
```
# When running "make menuconfig", enable "extra low-level
# configuration setup", select the 20KiB bootloader, disable "USB for
# communication", and select USART1 for the "Serial Port".
```

Salkio, 2021-01-30:
> Would you be so kind to consider reading the damned config file?
> 
> Quote:
> 
> ```
> # When running "make menuconfig", enable "extra low-level
> # configuration setup", select the 20KiB bootloader, disable "USB for
> # communication", and select USART1 for the "Serial Port".
> ```

As I sent that message and looked at the quote you sent and realized its all there. I do apologize!

closed 2021-02-28 by github-actions[bot]