Klipper3d/klipper #3865: timeout in [idle_timeout] is ignored

DasBootU96, 2021-01-30:
Weird behavior with Klipper:

When I restart Klipper service and then turn on heater bed only without touching anything else, the timeout in [idle_timeout] is ignored and the heater bed remains on. 

If you were to home the printer, the heaters will be automatically turned off after the specified timeout sine the last move.

When I restart Klipper service and perform a Home (G28) and then turn on heater bed without touching anything else, the timeout in [idle_timeout] is honored and the heater bed is turned off exactly as expected.

Used a 120 second timeout to test:

`[idle_timeout]
timeout: 120
`

Sometimes I turn on the heated bed only for it to heat the chamber, then I sometimes get distracted and forget about it. Expected behavior is for klipper to turn off heaters automatically after the timeout expires without having to 'move' (e.g. home) the printer

Printer is a Voron V0 cxy with SKR Mini E3 2.0 running Mainsail on Rpi3b+. Behavior has been confirmed by at least one other Voron Discord member.

Can you please confirm behavior, whether or not it is function as designed or bug or if there are workarounds available?

Thanks a lot.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5898536/klippy.log)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/5898537/printer.cfg.txt)

natnikos, 2021-02-02:
I can confirm the above behavior!

DasBootU96, 2021-02-04:
FWIW: Adding this [delayed_gcode](https://www.klipper3d.org/Command_Templates.html#delayed-gcodes) is a workaround:

`[delayed_gcode my_delayed_gcode]`
`gcode:`
  `  M119`
`initial_duration: 1`

Instead of `M119`  as gcode, also tried `M117 Hello` and `M114` without success.

Still weird though to have to do this.

modmt, 2021-02-07:
I have the same problem with SKR 1.4 and Mainsail on Rpi3B.

gcormier, 2021-02-23:
Same issue on klipper 0.9.1-259 and SKR1.4turbo.

wlhlm, 2021-02-24:
I can confirm this behavior on a Anet 1.4 controller board. I've used this behavior to pre-heat the printer during initial startup, so that I still have the timeout kick in once I've printed something, but can pre-heat for as long as I want). But I guess it's bug.

KevinOConnor, 2021-03-05:
To the best of my knowledge, the klipper code works as intended.  As a guess, some external process or delayed_gcode is continually issuing commands causing the idle_timeout to be reset.

Given our updated policy on github issues (see https://www.klipper3d.org/Contact.html ), I am closing this ticket.

-Kevin

wlhlm, 2021-03-05:
I'm hitting this issue with the Anet A8 config supplied with Klipper. There is no delayed_gcode in there.

DasBootU96, 2021-03-05:
Hi Kevin,

More and more people are running into this issue. No delayed_gcode is used. As for an external process, I would not know where to look. Just defaults are used and nothing special is configured.

Here are more logs and config files from the following reproducable actions using a 120 second timeout:

1. Turn on printer.
2. Set bed temperature to 50 (SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET=50)
3. Waited 8, 9 minutes, but bed does not turn off. 
4. Then home (G28) the printer and the bed turns of after 120 seconds.

[moonraker.conf.txt](https://github.com/KevinOConnor/klipper/files/6093609/moonraker.conf.txt)
[moonraker.log](https://github.com/KevinOConnor/klipper/files/6093610/moonraker.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/6093611/printer.cfg.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/6093612/klippy.log)
[mainsail.cfg.txt](https://github.com/KevinOConnor/klipper/files/6093613/mainsail.cfg.txt)

Can you please take another look or inform me on what additional information you might need?

Thank you.

wlhlm, 2021-03-05:
To confirm it with a log, I've updated Klipper to commit 1950380d6c90e57f6fa844f886d208e5391ede7b and freshly flashed the A8 board. I copied over the example configuration included in Klipper and ran the procedure @DasBootU96 outlined above (connected via Octoprint 1.5.2), only that I waited about 15 minutes in step 3 as the default Klipper config has a 600 second timeout. After issuing `G28` the bed heater is turned off after about 600 seconds, but just heating the bed after startup ignores that.

[klippy.log](https://github.com/KevinOConnor/klipper/files/6094042/klippy.log)

KevinOConnor, 2021-03-06:
@Arksine was able to diagnose the likely root cause of this issue (M109 command was not notifying "printer activity").  Should be fixed now (commit 64b3e564).

-Kevin

DasBootU96, 2021-03-06:
Updated klipper (v0.9.1-296) and tested: the heater bed now turns off automatically after the specified timeout without having to 'move' the printer first.

Thanks a 1000000! @Arksine and @KevinOConnor

closed 2021-03-06 by DasBootU96

evandepol, 2021-12-04:
Of note, i have a few control loops (using delayed_gcode) that update case leds automatically (based on activity) and also updated fans automatically (based on temperature in the electronics bay), and these too appear to prevent the idle_timout from triggering.

RoGueMS, 2022-04-01:
> This seems to be the case also after a FIRMWARE_RESTART, as idle timeout does not seem to trigger unless the axis of the printer are moved.
> 
> The timeout settings are these:
> 
> ```
> [idle_timeout]
> timeout: 10
> gcode:
>   M84
>   TURN_OFF_HEATERS
>   STATUS
> ```
> 
> After a firmware restart sitting in Idle for 10 seconds does nothing, as soon as I home the printer and leave it idle for 10 seconds it does execute the STATUS command.
> 
> When turning it on the status command returns the state "ready", which makes me think this might be intended behavior, however I find that this is different from what Is understood from the documentation [In the config reference](https://www.klipper3d.org/Config_Reference.html?h=idle+timeout#idle_timeout) and [in the status reference](https://www.klipper3d.org/Status_Reference.html?h=idle#idle_timeout) documentations, as, per my understanding, any "Ready" printer should go into "Idle" state after the time set in [idle_timeout] block in the config. [klippy.log](https://github.com/Klipper3d/klipper/files/7419703/klippy.1.log)

Sorry for the revive, but i found myself with the same issues. I finally figured out that the printer when it first turns on its already in "IDLE" mode so it wont trigger anything. But if you move it, do a g28, or raise the temps then it will be out of the IDLE mode flagging the timer to start until it idles again!

teeedubb, 2022-05-25:
Hi,

I am noticing the same behaviour - idle timeout is not triggered unless there is some interaction with the printer. Steps to replicate is to power on the printer + pi and let it sit with no interaction - the fluidd ui shows the printer as being idle, but it will idle indefinitely and does not shut down as it is configured to do. I do not have any delayed macros running on boot. With the log files below, the printer has been on for an hour.

[moonraker.txt](https://github.com/Klipper3d/klipper/files/8766992/moonraker.txt)
[klippy.txt](https://github.com/Klipper3d/klipper/files/8766993/klippy.txt)