Klipper3d/klipper #3866: Loosing connetion-reboot required

markwinger, 2021-01-30:
This problem has been giving me grief for a while.   I have been running octoprint, but recently changed to fluidd,  The problem exists on both so I believe it is klipper.

What happens is (usually when I restart firmware after config change) connection is lost and will not be restablished.  On octoprint the restart firmware and restart host buttons are gray and dont respond and you cannot type anything into the console.  On fluid I type status in the console and get no response.
Rebooting does not clear the problem.  I must power down the printer.  My rpi3 is powered from printer so it requires a full shutdown-power off, reboot to get things working again.  

It would be nice to determine why the firware become unresponsive, but even a way to get things working with the power down would be very helpful.  I have never seen this happen during printing but has happened when canceling a print.

In the conversion to fluidd, I made many changes to config and the problem occurred quite ofen.   The weird thing is it does not happen all the time.  Some times I will restart the firmware many times before it happens.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5898562/klippy.log)

Sineos, 2021-01-31:
Duplicate. See #3558
Usually restarting the klipper service followed by a FIRMWARE_RESET is enough.

markwinger, 2021-02-04:
I have been trying to use the button for host restart.  This did not work on octoprint or fluidd.  I thought that was the same thing but apparently not.

From a command line running:
sudo service klipper stop
sudo service klipper start

does get things running again.  No firmware_restart is required after this for me.

closed 2021-02-04 by markwinger