Klipper3d/klipper #3868: Lost MCU

nunorosado1991, 2021-01-31:
Hello i have a bear i3 with original board from prusa. 

And i have this problem when is printing..
what i need to do ?

[klippy(1).log](https://github.com/KevinOConnor/klipper/files/5899859/klippy.1.log)
[moonraker.log](https://github.com/KevinOConnor/klipper/files/5899860/moonraker.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/5899863/printer.cfg.txt)

Lost communication with MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Arksine, 2021-01-31:
You have the dreaded usbserial bug.  Long story short, the Einsy Rambo has two MCUs, the primary one that runs the main firmware (Prusa Firmware, Klipper, etc), and a secondary MCU that handles USB communications.  The firmware Prusa ships on the second MCU is bugged.  It corrupts serial transfers, the corruption gets worse over time which eventually results in a dropped connection.

It can be resolved if you are comfortable attempting to program the MCU.  See [this for more information](https://github.com/PrusaOwners/mk3-32u2-firmware).

nunorosado1991, 2021-01-31:
Hello I have upgrade thi second firmware and now i think its ok. 

I will more more Gcode and i will write here if it was solved.

Thanks a lot. 

now gcode 2 rambo 0  :D

nunorosado1991, 2021-02-02:
Solved with upgrade firmware