Klipper3d/klipper #3869: Timeout on serial connect

pollonzo, 2021-01-31:
in attachement the klipper log file.
I checked the port by command /dev/serial/by-id/* 
[klippy (11).log](https://github.com/KevinOConnor/klipper/files/5900060/klippy.11.log)

Sineos, 2021-01-31:
Your klipper version is from 2018. Try updating to the current code-base and follow the documentation. The printer board seems not to be setup correctly.

`MCU 'mcu' config: RECEIVE_WINDOW=192 SOFT_PWM_MAX=256 SERIAL_BAUD=250000 ADC_MAX=1023 PWM_MAX=255 CLOCK_FREQ=16000000 MCU=atmega2560 STATS_SUMSQ_BASE=256`

ATMEGA2560?

pollonzo, 2021-01-31:
sorry, i send wrong log,
thanks for fast answer
[klippy.11.log](https://github.com/KevinOConnor/klipper/files/5900543/klippy.11.log)

pollonzo, 2021-01-31:
i try now also a MainsailOS, in attachement the last log
[klippy (19).log](https://github.com/KevinOConnor/klipper/files/5900608/klippy.19.log)

pollonzo, 2021-01-31:
i also use command ls "-l /dev" ...and test also it  ttyAMA0
[klippy (3).log](https://github.com/KevinOConnor/klipper/files/5900674/klippy.3.log)

Sineos, 2021-02-01:
For me it looks like your board is incorrectly flashed.
- make sure to use the correct config as basis: https://github.com/KevinOConnor/klipper/tree/master/config
  - Probably your best bet is the generic-mks-robin-nano.cfg
- make sure to follow the instructions on top of the config file
- you might have to adjust some pins. I don't know the mini nor the nano

pollonzo, 2021-02-01:
solved with this script ./scripts/update_mks_robin.py ./out/klipper.bin ./out/robin_mini.bin

Br

closed 2021-02-01 by pollonzo