Klipper3d/klipper #3872: RUMBA32 Flash and button press

karnhack, 2021-01-31:
I have a problem flashing the RUMBA32.
The instructions say:
_Press and hold the BOOT button, then use USB connect to the MKS RUMBA32. The motherboard enters DFU mode._

The configuration I'm trying to flash is this:
[image]
[image]

When rumba 32 is not in dfu mode, its idProduct is 0483:**5740**

[image]

When i press and hold the BOOT button the motherboard enters DFU mode. The address is not found.

[image]

When RUMBA32 is in DFU mode its idProduct has changed 0483:**df11** and when i try to flash via idVendor and idProduct it shows DFU error 

[image]

I am attaching the log file but it does not contain useful information 

[klippy.log](https://github.com/KevinOConnor/klipper/files/5900516/klippy.log)

Thanks for your kind help

Gianluca

closed 2021-02-01 by karnhack

whiterabbit62, 2021-02-04:
Hello, did you fix this problem ?

I also have a btt rumba32 board, i am trying to make it work with klipper. Do you have a working config file assigned with pin numbers ?

karnhack, 2021-02-05:
Hello, the problem was caused by the fact that I was using the usb 3 port. Now i have solved it! Thanks

whiterabbit62, 2021-02-05:
> 
> 
> Hello, the problem was caused by the fact that I was using the usb 3 port. Now i have solved it! Thanks

Is there any way you could share your config file ?

karnhack, 2021-02-05:
> > Hello, the problem was caused by the fact that I was using the usb 3 port. Now i have solved it! Thanks
> 
> Is there any way you could share your config file ?

Hi, Pay attention because I have connected the drivers in spi

[printer.zip](https://github.com/KevinOConnor/klipper/files/5931923/printer.zip)

whiterabbit62, 2021-02-05:
Thank you so much. I will try to make it work with a known config now.

whiterabbit62, 2021-02-05:
Sorry, it's me again. May i ask how did you make the connection of Pi with Rumba32 board physically ?

I connected Rumba32 microusb to raspberry pi Zero'usd via micro USB to micro USB cable.

I am able to see the board when it is not in DFU mode;

pi@mainsailos:~ $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-STMicroelectronics_RUMBA32_F446VE_CDC_in_FS_Mode_359638753039-if00

karnhack, 2021-02-05:
Setting to DFU mode:
USB connect MKS RUMBA32, press and hold the "BOOT" button, then press the "RESET" light strip for more than 1S, release "RESET", and then release "BOOT".

When the rumba32 is in dfu mode this address not work: /dev/serial/by-id/usb-STMicroelectronics_RUMBA32_F446VE_CDC_in_FS_Mode_359638753039-if00

use usb id of your rumba : make flash FLASH_DEVICE=0000:1111

whiterabbit62, 2021-02-05:
i just realised that there are some differences between mks rumba32 and Btt rumba32. Mine is BTT rumba32.
I tried that command after getting in DFU mode;
make flash FLASH_DEVICE=0000:1111
it didnt work.

Do you have any information about BTT rumba32 8mhz or 12Mhz oscillator , maybe i am setting wrong make menuconfig options.

karnhack, 2021-02-05:
0000:1111 it was just one example.
Get the ID address with the command: `lsusb`  or `lsusb -v` (don't use usb3 on your raspberry)

[image]

The ID address must also be entered when you send the command `make menuconfig`
Select: Use USB for communication

[image]

then enter the ID you got with the lsusb command 

[image]

Oscillator is 12Hz

whiterabbit62, 2021-02-05:
I made it! after installation of klipper, is it normal that i cant see the board with lsusb command again ?

Is it same for you too ?

karnhack, 2021-02-05:
roboot all

whiterabbit62, 2021-02-05:
I rebooted but still no chance. I have 2 BTT rumba32 boards, i flashed klipper both of them and after installation, 

pi@mainsailos:~ $ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Before installation i was able to see ;
0483:5740 normal boot,
0483:df11 DFU

I am curious if it's same situation for you too.

karnhack, 2021-02-05:
In my case the ID 0483:5740 appeared because the usb was connected to usb3 (the blue port) 
When I connected the cable to usb2 (the black port) the two ids (normal and dfu) were the same.

[image]

In the usb ID you have to enter 0x0483 and 0xdf11

karnhack, 2021-02-05:
Turn off the rumba32 (turn off the power supply and disconnect the cable from the raspberry). Restart the raspberry. Plug the rumba32 back in and try again. It happened to me too

whiterabbit62, 2021-02-06:
I guess one board has bricked bootloader, even windows do not see it unless it is in DFU mode.
The other board i can see it both in DFU and normal mode after lsusb command but the other one has something wrong.

Is there any way i can flash the bootloader and bring it back ?

i managed to flash marlin via platformio and it flashed succesfully so board is still alive. I also have ST-link to program. Just couldnt find any guide to follow for flashing bootloader.

whiterabbit62, 2021-02-06:
it's so weird. Before i was about to give up one of the board that was not being able to recognized by PC(unless DFU mode) and also pi zero (Not visible in both DFU and normal mode), This morning i flashed problematic board with Marlin via platformIO again.
Windows started to see it, then i checked it on pi zero,

Normal boot
pi@mainsailos:~ $ lsusb
Bus 001 Device 005: ID 0483:5740 STMicroelectronics Virtual COM Port
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

DFU boot,
pi@mainsailos:~ $ lsusb
Bus 001 Device 006: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Then i flashed klipper successfully. Now they are all working flawlessly.

Thank you so much for you patience, Hope this helps others who are having same issues with their rumba32 boards.