Klipper3d/klipper #3874: Counterproductive startup order of operations with BLTouch/similar leads to oozing and ruined prints

dci87, 2021-02-01:
I've been very impressed with Klipper over the past few days I've been trying it out, but I've run into a nuisance that should be easily fixed. This is the procedure Klipper currently follows prior to executing any user gcode:

1) Heat bed, wait
2) Heat nozzle, wait
3) Home X/Y
4) Home Z with probe
5) Begin executing user gcode

If you don't have a height probe this works ideally and homing Z in the bottom-left corner even helps rub any accumulated ooze off where it's unlikely to affect anything. But if the user does have a height probe, what happens is:

1) Heat bed, wait
2) Heat nozzle which begins oozing, wait
3) Home X/Y
4) Home Z with probe, possibly depositing accumulated nozzle ooze right in the middle of the bed
5) Begin executing gcode with lots of accumulated ooze ready to rub off somewhere in the print if it wasn't already deposited right in the middle of the bed

It should be possible to correct this problem by changing the startup order of operations. If the user has a height probe then the nozzle isn't involved in the Z axis homing procedure, so there's no need for it to be heated so early:

1) Heat bed, wait
2) Home X/Y
3) Home Z with probe
4) Move nozzle back to X/Y home (this lowers the risk of any ooze reaching the table from landing anywhere important)
5) Heat nozzle, wait
6) Begin executing user gcode

This might be optimized to reduce startup time if possible:

1) Heat Bed (do not wait)
2) Home X/Y
3) Move probe to Z axis testing position
4) Wait for bed to reach full temperature
5) Heat Nozzle (do not wait, it should be safe to start now because it shouldn't reach oozing temperature until after the Z axis is homed)
6) Home Z with probe
7) Move nozzle to X/Y home
8) Wait until nozzle has reached full temperature
9) Begin executing user gcode

I don't know what the stock Marlin/Creality/whoever firmware behavior for the above procedure is but if they work the same way then fixing it would make another advantage for Klipper.

Some things I'm going to try as a workaround are extending the startup nozzle cleaning in Cura, adding additional skirt layers, or using a material that doesn't need to be heated as high, but these will add time and/or cost while still risking leaving a blob of plastic in the middle of the bed unless I babysit the printer during the startup procedure. Changing the firmware's startup order of operations seems like the best, most reliable, and most user-friendly way to solve this problem.

Thank you

BlackStump, 2021-02-01:
You really should read the Klipper Documents,  [Macro](https://www.klipper3d.org/Command_Templates.html)

dci87, 2021-02-01:
I didn't think a log file would be applicable because the software is working as intended but in a counterproductive way because of my less-common setup using a BLTouch probe. Here's one in case it's helpful: [klippy.zip](https://github.com/KevinOConnor/klipper/files/5903392/klippy.zip)

I have already looked through the Klipper Documents at length to the best of my very limited ability and I don't see an obvious section on modifying the printing startup order of operations. I apologize if I missed it. I'll look again tomorrow.

If I had the technical skill to correct this problem myself (I do not), that would solve it for one person. The issue I'm describing is not a user-specific preference that should be adjusted on the user's end, it's a functional issue affecting every user of this firmware whose printer has a Z axis probe and is therefore best corrected at the source. I do not have the skill to do that, so my options are to just ignore the problem or attempt to bring it to someone else's attention who can fix it.

Earlier tonight I saw a problem with a simple solution. I looked through the documentation for anything relating to changing the firmware's startup procedure, I spent an hour writing clear instructions on how to reproduce the problem and offered constructive ideas for how the problem might best be solved. I went to the contact page, followed the steps, made a futile attempt on discord where people are currently discussing the sorts of things you'd expect, and finally posted here.

The simplest form of this fix I propose should be as little moving a single line of gcode down a few lines in whatever file that's buried in. This is is the most I can offer.

Arksine, 2021-02-01:
> The issue I'm describing is not a user-specific preference that should be adjusted on the user's end, it's a functional issue affecting every user of this firmware whose printer has a Z axis probe [...]

The "startup sequence" is not defined by Klipper, it is defined by your slicer's "start gcode".  If you want to change the order of operations prior to beginning the print that is the place you need to look.

closed 2021-02-05 by github-actions[bot]