Klipper3d/klipper #3876: [FR] B-Spline (LPFBS) control

mrx23dot, 2021-02-01:
Engineers at the University of Michigan’s Smart and Sustainable Automation Research Laboratory (S2A Lab) have landed an algorithm capable of keeping layers in check at up 10 times the typical speed of an FFF 3D printer.

Please read
https://3dprintingindustry.com/news/cut-fff-3d-print-times-half-university-michigan-123158/

We have the computing power, not sure if it's useful enough, pls comment.

related:
https://github.com/MarlinFirmware/Marlin/issues/8071
https://github.com/MarlinFirmware/Marlin/issues/8308

dmbutyugin, 2021-02-01:
Klipper supports a different algorithm - input shaping. Well, in a broad sense their LPFBS is also an input shaping technique, Klipper simply supports different, simpler input shapers. They also allow one to bump the acceleration and reduce the print time. More importantly, they allow simple manual and automatic calibration by the user. In principle, it might be possible to implement LPFBS, because Klipper will likely have enough computational power, but the calibration is still an open question.

mrx23dot, 2021-02-01:
By the looks of it B-spline is only a PT1 low pass filter that smooths out movement between current point and next 2 points.

https://en.wikipedia.org/wiki/B-spline

dmbutyugin, 2021-02-01:
Yes, but it does not mean that S2A Lab folks use the points between G0/G1 moves as reference points for their B-Splines. Instead, it seems that they have pretty elaborate method to determine "control points" in their article to minimize the positioning error.

mrx23dot, 2021-02-01:
I read through the paper, so they compared G1 (linear) / G2_3 (arc) / G5 (cubic spline)  movements.
G1 causes resonance because of sharp movements
G2_3 were fastest (arcwelder already converts from G1 to these)
G5 is 2nd fastest, but most precise and smallest gcode  https://marlinfw.org/docs/gcode/G005.html

Sounds like it's a pre-processing step on gcode before printing (just like arcwelder)
Is it worth pre-processing the gcode, or input shaper has the same effect?

dmbutyugin, 2021-02-01:
Are you talking about this article
https://www.sciencedirect.com/science/article/pii/S2351978919311059
by any chance?

Because the other one, referenced in that other article
https://www.sciencedirect.com/science/article/abs/pii/S0957415817301277
talks about LPFBS schema (limited-preview filtered B-spline).

mrx23dot, 2021-02-01:
Yeah first one.

Their LPFBS:
http://www-personal.umich.edu/~molong/pdf/pub/VMPT17_62.pdf
in detail: (tries to offload calculation to cloud to sell it as a service), Table 3 has **nice comparison pics**
https://res.mdpi.com/d_attachment/inventions/inventions-03-00056/article_deploy/inventions-03-00056-v2.pdf

Prove me wrong but limited-preview filtered B-spline (LPFBS) is just a fancy way saying they are using G5 commands.

dmbutyugin, 2021-02-01:
> Prove me wrong but limited-preview filtered B-spline (LPFBS) is just a fancy way saying they are using G5 commands.

Sure, that's accurate. Their know-how is how to compute the parameters for those G5 commands from the original G0/G1 commands using LPFBS algorithm (e.g. (1) and (2) in the linked article).

mrx23dot, 2021-02-01:
If it's in the marlin manual than it cannot be that hard, even with random trying parameters :)
Added an extra patent above.

The question, does it add anything extra to the current input shapper or it's similar.

dmbutyugin, 2021-02-02:
> If it's in the marlin manual than it cannot be that hard, even with random trying parameters :)

G5 implementation isn't hard. But I doubt it is easy to generate those splines from the input GCode, based on the article that I briefly read.

> The question, does it add anything extra to the current input shapper or it's similar.

Great question. The current input shapers in Klipper may add some smoothing (depending on the lowest resonance frequency of the printer). They claim their approach also eliminates ringing and allows faster printing, but they don't cover what kind of defects (if any) their method adds instead. They also don't mention how robust their algorithm is to the changes in the printer configuration - e.g. as the large model is printed on a bed slinger printer, the mass of the bed changes, and so does the frequency response. Input shapers perform acceptably in this case (and it is possible to tune them to work well in both cases - with empty bed and with a heavy print). But I could have missed that in the articles, since I didn't read them very attentively.

mrx23dot, 2021-02-02:
As I read Klipper's input shapper only adjusts acceleration profiles and speed with look ahead of upcoming points, but not the points itself.
Rounding sharp corners/movements (modifying points) would make it possible to step up speed (with aesthetic cost).

Anyway, looks like this should be done outside Klipper.
Conversation continues at https://github.com/FormerLurker/ArcWelderLib/issues/33

closed 2021-02-02 by mrx23dot

ProfessorChi, 2021-06-13:
> G5 implementation isn't hard. But I doubt it is easy to generate those splines from the input GCode, based on the article that I briefly read.
>
>
> Great question. The current input shapers in Klipper [...]

@dmbutyugin First off, I'm very grateful to the Klipper team for bringing input shaping to the 3D printing community. As you probably already know, it has been available in CNCs for a long time and one of its biggest weaknesses is the corner rounding and smoothing problem which you reference. This was a main motivation for us working on the FBS algorithm, which does not cause corner rounding. You can see a comparison here: https://umich.box.com/s/pct75ugrcow90bnxpjgjf28cnnfxwv4n . That said, input shaper is much simpler to implement compared to FBS, and does not require as much computation. The calibration of input shaper is also much simpler than FBS. 

From our several tests, the FBS algorithm shows remarkable robustness. Input shaping adds robustness at the expense of more corner rounding. We are currently developing an even more robust version of the FBS algorithm that does not introduce rounding. It is still in its academic stage: https://umich.box.com/s/7tuqmu8t33bebyqg66x5kesdkx0ex915 but we are now working to make it more practical. The FBS algorithm also has the ability to compensate machines with position dependent resonances - like delta 3D printers - by automatically altering its compensation as the printer moves. We are working on this algorithm. Input shaping does not have this property.

We'll soon be sharing a whitepaper comparing input shaper and FBS but I appreciate your comments that have helped people see that they are not the same algorithm, though they are seeking to achieve similar goals.