Klipper3d/klipper #3878: Fanspeed does not responde according to settings.

marioy0, 2021-02-02:
Hello,

i am using an Ender3v2 with the 4.2.2 mainboard. I upgraded the fanduct to use two 4020 Sunon Meglever fans. 
Also i use fluidd on an Raspberry 3b+. 

My Problem is now that the fan stop speening below S255. I want to print some parts with 20% - 50% Fan speed but cant get it to work. Disableing the fan completty isnt an option as the Mainboard Fan is linked to the Partcooling Fan. So the mainboard would heat up. 

I have readed that the Problem is with the frequency of the Fan, but i dont know how to change it in Klipper. Also is their an Option to tell klipper to split the Mainboard Fan and Partcooling Fan ? ?

marioy0, 2021-02-02:
the klippy log
[klippy.log](https://github.com/KevinOConnor/klipper/files/5910233/klippy.log)

jorkkiporkki, 2021-02-02:
The Sunons do not work with software PWM, they require hardware PWM at around 20 kHz to work properly
Unfortunately that is not supported on STM32-architechture yet, hopefully one day, See #3808.
Some add small capacitors across the fan, but I don't recommend that because of the high currents and EMI issues. At least you should add a resistor of a few ohms in series with the fan (with proper wattage).

The idea to connect the partcooler and cpu cooler in parallel is a Creality brain fart.
 Unless you have the external electronics to control the cpu cooler via a pin (e.g. PA13 alias SWDIO), you should rewire the cpu cooler to the FAN screw terminals on the board. That way the cooler will run constantly as is the hotend cooler.

marioy0, 2021-02-02:
ok thank you for the information. can you recommend another 4020 fan with software pwm, that is similar to sunon ?

jorkkiporkki, 2021-02-02:
Sorry, haven't used any in that size. I'm sure you get plenty of recommendations on the Discord: https://discord.com/invite/df9yTE2

closed 2021-02-05 by github-actions[bot]

hcgonzalezpr, 2021-11-10:
@marioy0 The following fan settings will enable 25% to 100% adjustment similar to Marlin is able to do for the Sunon:

**off_below: 0.25**
**cycle_time: 0.7**

pleuriskind, 2021-12-15:
> @marioy0 The following fan settings will enable 25% to 100% adjustment similar to Marlin is able to do for the Sunon:
> 
> **off_below: 0.25** **cycle_time: 0.7**

will that works? i have the same fan and problem!