Klipper3d/klipper #3882: [enhancement] Add relative mode support for G2/G3

FormerLurker, 2021-02-03:
I've received an issue recently for my arc welder plugin (converts G0/G1 to G2/G3) about the following klipper error message:

```
G2/G3 does not support relative move mode
```

I tracked down the message to this code:

https://github.com/KevinOConnor/klipper/blob/d9af6129eacacefa6b33d6ed45a593ef6e11fddb/klippy/extras/gcode_arcs.py#L27-L28

It seems like it wouldn't be too difficult to add support for this.  I and J are always relative, so it would just be a matter of calculating the absolute coordinates of X, Y and Z.  I'm willing to submit a PR, but at the moment I'm building a new printer, so it would be a while before I can load up klipper and run some tests.  However, I would eventually get to this in a month or two.

Please let me know what you think.

FormerLurker, 2021-02-03:
Ok, I'll close this issue since I now see feature requests are not allowed (though I think this one is bordering on a bug since this is truly unexpected behavior, and differs for all other firmware I'm aware of).  

I did also want to point out that I don't believe the original 'how to post an issue' link explained this.  I got a different link when I created this as a, 'first time poster', and that message has changed, so I can't verify this unfortunately.  I did navigate to the link, and I did read the text, but I'm known to miss stuff from time to time, so sorry about that..

KevinOConnor, 2021-02-03:
If you're working on G2/G3 support (including working on it, by researching the changes) then, in my opinion, it's okay to open a ticket.

FWIW, it hasn't been implemented because no one has gotten to implement it yet.  I'd guess that if you get to it in a month or two, you'll still be the first person to get to it.

-Kevin

FormerLurker, 2021-02-03:
Excellent, thank you!  I was getting strong 'do not disturb' feelings, and was going to abandon this, so I really appreciate your response.  I totally get being extremely strict about things like this (I get flooding as well).  I will reopen.  Let me know if I should change the title.

I think since the changes will likely only be 3 lines of code or so and removal of the original error message, that I can take this on myself and farm out testing to the folks who complained about the error message in my repo.

Thanks again!

ETE-Design, 2021-02-07:
@FormerLurker Would be great if you can implement G2 / G3 to Klipper :-)

FormerLurker, 2021-02-07:
@ETE-Design, to be clear, it is already implemented, it just doesn't work in relative mode.  I've been reviewing the algorithm, and it looks like it was taken from Marlin 1.x, and is missing some of the advanced (and extremely useful) parameters, like min-arc-segments.  I'm going to add that too, if possible, since it makes the default mm-per-arc-segment of 1.0mm work well, even for very tiny arcs (below 5mm).  Since Klipper is nearly immune to slowdowns when using lower values for mm-per-arc-segments, I also suggest the default for that parameter be lowered somewhat, which may improve surface finish on larger arcs.

FormerLurker, 2021-03-01:
I think this should stay open a bit longer.  Just taking a while for me to get around to a resolution, and it will be useful to include the issue # with the PR.

closed 2021-03-30 by github-actions[bot]

moebis, 2021-05-07:
@FormerLurker this was a great idea and plan. Is it abandoned now? I wanted to test the Arc Welder plugin in Cura.

FormerLurker, 2021-05-07:
@moebis, I still hope to get to this, but I am having some hardware issues.  I'm currently building the printer I want to run with Klipper, and am working out some issues.  Will reopen when I get to it.

moebis, 2021-05-07:
@FormerLurker Thanks! let me guess... a Voron v2.4? I started sourcing parts for mine a couple of week ago. I'm running your Arc Welder right now, didn't realize I just had to enable the "[gcode_arcs]" in the printer.cfg ....I'm running a tolerance coin right now, amazed at how small the file size is and how smooth it's running. This should be on by default. When I first ran it Klipper didn't understand G2/G3.