Klipper3d/klipper #3886: [POC] DHT temperature sensor and Hardware PWM on Raspberry Pi 4B

cinadr, 2021-02-03:
Hi!

I managed to make my 12V NZXT fan connected to PWM GPIO to be regulated by a DHT22 sensor also connected with GPIO through overlays. 
12 V fan is connected like this: 
[image]
My DHT is connected like this - on pin 16:
[image]
This is my config.txt relevant lines:
```
dtoverlay=pwm,pin=12,func=4
dtoverlay=dht11,gpiopin=16
```
To init pwm before klipper starts I needed a startup script for my fan:
```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
import time
import os
import sys

PERIOD = 75000

def write(value,path):
  os.system("echo {v} > {p}".format(v=value,p=path))

def init_pwm():
  if not os.path.isdir("/sys/class/pwm/pwmchip0/pwm0"):
    print("Initializing PWM functions")
    write("0","/sys/class/pwm/pwmchip0/export")    # pin12 is on pwmchip0, we need to export that
    time.sleep(2)                                                        # wait for export to be done
  print("Setting period length")
  write(PERIOD,"/sys/class/pwm/pwmchip0/pwm0/period")

def set_fanenable(enable):
  print("Enabling fan" if enable else "Disabling fan")
  write("1" if enable else "0","/sys/class/pwm/pwmchip0/pwm0/enable")

if __name__ == "__main__":
  init_pwm()
  set_fanenable(1)
```
Put it in crontab to fire after reboot:
```
# chrontab -e
@reboot /home/pi/fanstart
```
I compiled mcu for raspberry: [see here](https://www.klipper3d.org/RPi_microcontroller.html)
Created a config with this [include raspi.cfg] in printer.cfg:
Pin needs to be set to pin0 (no matter what pin used in overlay settings as pwmchip0 is needs to be found by klipper.
The relevant macro is HARD_PWM_TO_CHIP and HARD_PWM_TO_PIN in the file:
[https://github.com/KevinOConnor/klipper/blob/d9af6129eacacefa6b33d6ed45a593ef6e11fddb/src/linux/hard_pwm.c](https://github.com/KevinOConnor/klipper/blob/d9af6129eacacefa6b33d6ed45a593ef6e11fddb/src/linux/hard_pwm.c)

```
[mcu host]
serial: /tmp/klipper_host_mcu

[temperature_sensor rpi_temp]
sensor_type: rpi_temperature

[temperature_fan enclosure_fan]
pin: host:gpio0 
sensor_type: dht_temperature
hardware_pwm: true
max_power: 1.0
shutdown_speed: 0
kick_start_time: 0.1
off_below: 0.4
cycle_time: 0.000075
control: pid
pid_Kp: 3
pid_Ki: 3
pid_Kd: 5
target_temp: 20.0
max_temp: 40.0
min_temp: 10.0
max_speed: 1.0
min_speed: 0.5
```
Created a dht_temperature.py file in ~/klipper/klippy/extras based on rpi_temperature.py.
Error checking needed some tweak as almost every second read throws an IOError so we give back previous read results.
```
# Support for Raspberry Pi DHT temperature sensor
#
# Copyright (C) 2020  Al Crate <al3ph@users.noreply.github.com>
#
# This file may be distributed under the terms of the GNU GPLv3 license.

import logging

RPI_REPORT_TIME = 1.0
PROC_TEMP_FILE = "/sys/bus/iio/devices/iio:device0/in_temp_input"

class DHTTemperature:
    def __init__(self, config):
        self.printer = config.get_printer()
        self.reactor = self.printer.get_reactor()
        self.name = config.get_name().split()[-1]

        self.temp = self.min_temp = self.max_temp = 0.0

        self.printer.add_object("dht_temperature " + self.name, self)
        if self.printer.get_start_args().get('debugoutput') is not None:
            return
        self.sample_timer = self.reactor.register_timer(
            self._sample_dht_temperature)
        try:
            self.file_handle = open(PROC_TEMP_FILE, "r")
        except:
            raise config.error("Unable to open temperature file '%s'"
                               % (PROC_TEMP_FILE,))

        self.printer.register_event_handler("klippy:connect",
                                            self.handle_connect)

    def handle_connect(self):
        self.reactor.update_timer(self.sample_timer, self.reactor.NOW)

    def setup_minmax(self, min_temp, max_temp):
        self.min_temp = min_temp
        self.max_temp = max_temp

    def setup_callback(self, cb):
        self._callback = cb

    def get_report_time_delta(self):
        return RPI_REPORT_TIME

    def _sample_dht_temperature(self, eventtime):
        try:
            self.file_handle.seek(0)
            currtemp = float(self.file_handle.read())/1000.0
        except Exception:
            #logging.exception("dht_temperature: Error reading data")
            if self.temp < self.min_temp:
               self.temp = self.min_temp
            currtemp = self.temp
            #return self.reactor.NEVER

        self.temp = currtemp
        if self.temp < self.min_temp:
            self.printer.invoke_shutdown(
                "DHT temperature %0.1f below minimum temperature of %0.1f."
                % (self.temp, self.min_temp,))
        if self.temp > self.max_temp:
            self.printer.invoke_shutdown(
                "DHT temperature %0.1f above maximum temperature of %0.1f."
                % (self.temp, self.max_temp,))

        mcu = self.printer.lookup_object('mcu')
        measured_time = self.reactor.monotonic()
        self._callback(mcu.estimated_print_time(measured_time), self.temp)
        return measured_time + RPI_REPORT_TIME

    def get_status(self, eventtime):
        return {
            'temperature': self.temp,
        }

def load_config(config):
    # Register sensor
    pheaters = config.get_printer().load_object(config, "heaters")
    pheaters.add_sensor_factory("dht_temperature", DHTTemperature)
```
Modified def setup_sensors(self, config) code of heaters.py in same folder to include new dht_temperature sensor:
```
    def setup_sensor(self, config):
        modules = ["thermistor", "adc_temperature", "spi_temperature",
                   "bme280", "htu21d", "lm75", "rpi_temperature","dht_temperature",
                   "temperature_mcu", "ds18b20"]
```
Executed:
```
#  ~/klippy-env/bin/python -m compileall dht_temperature.py
#  ~/klippy-env/bin/python -m compileall heaters.py
```
And at the end reboot RPI.

I've:
[image]

Can this be put in klipper somehow? I'm not an experied github user so might someone make a pull request based on the above? (Lot of changes involved all around the code and some help/samples need to be written as well)

KevinOConnor, 2021-02-28:
Thanks.  We do take code contributions for Klipper, but it's best to do that using a github Pull Request.  There's a little bit of info on that at https://github.com/KevinOConnor/klipper/blob/master/docs/CONTRIBUTING.md .

-Kevin

cinadr, 2021-03-02:
Hi!

I managed to get this working since [this](https://github.com/KevinOConnor/klipper/commit/a1baefc2c2da542877809deefef73321670eaa5e)  klipper commit. So no Pull Request is necessary.

The wiring to my Pi is the same as above. I have a proxy service that reads DHT temperature data as it gives back a lot of I/O and timeout errors and write current value to a file. This script also initializes my PWM for fan control.
You can change TEMP, HUMID, INTERVAL values to your liking. But it is important to set TEMP path in klipper.cfg  (see later). Here is the code:
```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
import time
import os
import sys
import signal
import getopt
import logging

logging.basicConfig(format='%(asctime)s %(levelname)-8s %(message)s', filename='dht_proxy.log', level=logging.INFO)

TEMP     = '/home/pi/dht_temp'
HUMID    = '/home/pi/dht_humid'
INTERVAL = 10 # seconds

def write(value,path):
  try: 
    os.system("echo {v} > {p}".format(v=value,p=path))
  except:    
    logging.error("Error writing {v} to {p}. Error is {e}".format(v=value,p=path,e=sys.exc_info()[0]))
  else:
    logging.debug("Writing {v} to {p}".format(v=value,p=path))

def read(path):
  f=open(path, "r")
  error=0
  while True:
    try:
      value = float(f.read())
    except OSError:
      error+=1
      if error < 10 :
        continue
      else:
        logging.error("Last 10 reads of {p} is unsuccessful. Skipping this.".format(p=path))
        value = 0.0
        break
    break
  f.close()
  return value

class Fan(object):
  def __init__(self):        
    self.fan_max  = 100
    self.fan_off  = 0
    self.fan_on   = 70  # Make sure to turn on at this percent
    self.fan_high = 99
    self.fan_low  = 40
    self.divider  = 750
    self.period   = 75000
    self.fan      = 0
    self.fanenabled=True    
    self.__initpwm()
    self.__enablefan()

  def __initpwm(self):
    if not os.path.isdir("/sys/class/pwm/pwmchip0/pwm0"):
      logging.info("Initializing PWM functions")
      write("0","/sys/class/pwm/pwmchip0/export")
      time.sleep(2)
    if not os.path.isdir("/sys/class/pwm/pwmchip0/pwm0"):
      logging.error("Cannot initialize pwm chip. Exiting.")
      sys.exit("Cannot initialize pwm chip. Exiting.")
[... 89 lines cut ...]
sensor_type: temperature_host                                  
sensor_path: /sys/class/thermal/thermal_zone0/temp             
                                                               
[temperature_fan Enclosure]                                    
pin: host:gpio0                                                
sensor_type: temperature_host                                  
sensor_path: /home/pi/dht_temp                                 
hardware_pwm: true                                             
max_power: 1.0                                                 
shutdown_speed: 0                                              
kick_start_time: 0.1                                           
off_below: 0.4                                                 
cycle_time: 0.000075                                           
control: watermark                                             
#pid_Kp: 3                                                     
#pid_Ki: 3                                                     
#pid_Kd: 5                                                     
max_delta: 2.0                                                 
target_temp: 25.0                                              
max_temp: 60.0                                                 
min_temp: 0.0                                                  
max_speed: 1.0                                                 
min_speed: 0.0                                                 
```
I don't know how to use the PID_CALIBRATE with this or is it even possible as I don't have any direct control on enclosure temperature. It's just influenced by the temp of BED and NOZZLE in a very long time (30-60 secs or more, depends on enclosure size and spontaneous ventilation). So I changed to watermark control. With this the fan is operated on/off type and keeps temperature around 25 °C.

closed 2021-03-26 by cinadr