Klipper3d/klipper #3887: Need help for stm32h7 port

D4SK, 2021-02-03:
I'm currently trying to port klipper to stm32h750 mcus and running into a "Rescheduled timer in the past" error.

I think I have most of it implemented, whats missing is SPI, USB and ADC control. I haven't looked much into the timer and irq handeling aspect, so this could be the problem. All the abstractions armcm_timer needs are provided by the cmsis library, and startup/communication works fine, so not sure where to start there. timer_read_time seems correct as well.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5921796/klippy.log)
[klippy_shutdown00037.log](https://github.com/KevinOConnor/klipper/files/5921797/klippy_shutdown00037.log)
I already tried running at 200mhz instead of 400, and increasing the timeout in timer_dispatch_many from 1000us to 1000000us, both created the same error.
When connecting with console.py get_config returns is_shutdown=1

What could cause this, are there any other irregularities in the log I should look at?
Other than the clocks, are all times in the log in seconds?

https://github.com/D4SK/klipperui
This is the repo I'm working on (master branch is based on work-python3), it has some custom host plugins and a few modifications going along with that. Nothing affecting the mcu though.

Thanks for any help, I like the project.

KevinOConnor, 2021-02-05:
Interesting.

If you're looking for my review, then my feedback is that there are too many changes on that branch for me to review it.  Best would be to put together a branch with a small number of commits that just add the core stm32h7 support, make sure it passes the existing regression test cases, and open a PR with that branch.  That will make it easier to review and spot any problems.

-Kevin

D4SK, 2021-02-09:
No problem, will open a PR tomorrow.

closed 2021-02-09 by D4SK