Klipper3d/klipper #3893: Internal error on command:"G28" after upgrade con 0.9.1

svkoni, 2021-02-06:
I got the above mentioned Failure after upgrading from v 0.7 to 0.9.1. I then downgraded to 0.8. after that the G28 command worked as expected. Upgraded again, same problem. On the other hand, The rotation_distance: and microsteps: in the config were not accepted but the step_distance: is still required. i guess that points the problem. The STEPPER_BUZZ commands were succesful. Might be something in the upgrade process went wrong? 

Thanks for your help

Sven

[klippy.log](https://github.com/KevinOConnor/klipper/files/5936831/klippy.log)

FirewinX, 2021-02-15:
i have the same error on a SKR 1.4 with klipper v0.9.1-229. on homing command crashes the mainboard.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5984152/klippy.log)

KevinOConnor, 2021-03-05:
The error appears to be the result of an incomplete software upgrade.  See the FAQ for the full software update sequence.

Given our updated policy on github tickets (see https://www.klipper3d.org/Contact.html ) I am closing this ticket.

-Kevin

closed 2021-03-05 by KevinOConnor