Klipper3d/klipper #3898: Unable to read tmc uart robin mini tmc2209 one wire

pollonzo, 2021-02-07:
Hi guys,
I try to configure klipper with my printer delta.
in attachement the printer.cfg klipperlog and jpg of my hw configuration.
If i install marlin UHCWNTPULR of this link ...
https://github.com/Foxies-CSTL/Marlin_2.0.x/tree/FLSUN_QQS-PRO-MULTI/Marlin/Firmwares

the printer work correctly..

[image]

[klippy (12).log](https://github.com/KevinOConnor/klipper/files/5938529/klippy.12.log)

https://pastebin.com/xVyZ3cZL

ymboc, 2021-02-07:
I have this board & printer on hand and am trying to get it working as well.  

Similarly  as a basis- point I have fully functional self-configured & compiled Marlin firmware based off of Foxies-CSTLs work as well, (I have some different hardware/thermistors and UART wiring that precludes me from flashing pre-compiled firmware)

I took a quick look at your klippy.log and noticed you're not yet applying the 'uart_address:' setting (and it's required)

For Reference, my TMC2209 sections: (note:  my uart_pin is physically different as are my addressing jumpers from your setup)

`[tmc2209 stepper_a]
uart_pin: PA9
uart_address: 0
microsteps: 16
interpolate: false
run_current: 0.9
hold_current: 0.8
sense_resistor: 0.110
stealthchop_threshold: 150

[tmc2209 stepper_b]
uart_pin: PA9
uart_address: 1
microsteps: 16
interpolate: false
run_current: 0.9
hold_current: 0.8
sense_resistor: 0.110
stealthchop_threshold: 150

[tmc2209 stepper_c]
uart_pin: PA9
uart_address: 2
microsteps: 16
interpolate: false
run_current: 0.9
hold_current: 0.8
sense_resistor: 0.110
stealthchop_threshold: 150

[tmc2209 extruder]
uart_pin: PA9
uart_address: 3
microsteps: 16
interpolate: false
run_current: 0.9
hold_current: 0.8
sense_resistor: 0.110
stealthchop_threshold: 30`

ymboc, 2021-02-07:
For reference I am still trying to my own TMC communication errors (_"Unable to read tmc uart 'stepper_a' register IFCNT"_) 

Is there a way to configure the Baud Rate for TMC communication? My working marlin configuration specifies "TMC_BAUD_RATE  19200" as it's the only potential configuration difference I have been able to identify between my Marlin & Klipper software configuration.

pollonzo, 2021-02-07:
uart_addess is also set on my printer.cfg under uart_pin.
I tried some stuff without success...
I will try tmc baudrate this evening.
Thanks so much

ymboc, 2021-02-07:
TMC_BAUD_RATE is a marlin setting; perhaps its not configurable on Klipper for good reason.

Recommend we discuss further in Klipper-Discord #stepper-drivers (or by DM) and try to keep it more formal here.

I don't think we can be sure yet that our difficulty isn't due to a mis-configuration (as opposed to a veritable bug/issue ).

closed 2021-03-09 by github-actions[bot]