Klipper3d/klipper #3900: Unable to read tmc uart 'stepper_X' register IFCNT

Techfluent-au, 2021-02-07:
Hi There looking for support for the Unable to read tmc uart 'stepper_X' register IFCNT

Big Tree Tech SKR1.3 
Big Tree Tech 2208 v3 

I have solder the pads on the bottom side of the driver, I have the jumpers in the correct location on the SKR.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5938959/klippy.log)

`#####################################################################
# 	X/Y Stepper Settings
#####################################################################

[stepper_x]
##	Connected to X on mcu_xye (B Motor)
step_pin: P2.2
dir_pin: !P2.6
enable_pin: !P2.1
rotation_distance: 40
microsteps: 16
full_steps_per_rotation:200  #set to 400 for 0.9 degree stepper
endstop_pin: P1.28
position_min: 0
position_endstop: 350
position_max: 350
homing_speed: 25   #Max 100
homing_retract_dist: 5
homing_positive_dir: true

[tmc2208 stepper_x]
uart_pin: P1.17
interpolate: True
run_current: 0.8
hold_current: 0.7
sense_resistor: 0.110
stealthchop_threshold: 0

[stepper_y]
##	Connected to Y on mcu_xye (A Motor)
step_pin: P0.19
dir_pin: !P0.20
enable_pin: !P2.8
rotation_distance: 40
microsteps: 16
full_steps_per_rotation:200  #set to 400 for 0.9 degree stepper
endstop_pin: P1.26
position_min: 0
position_endstop: 350
position_max: 350
homing_speed: 25  #Max 100
homing_retract_dist: 5
homing_positive_dir: true

[tmc2208 stepper_y]
uart_pin: P1.15
interpolate: True
run_current: 0.8
hold_current: 0.7
sense_resistor: 0.110
stealthchop_threshold: 0`

whiterabbit62, 2021-02-07:
I had same issue with twotrees TMC2209 V2 (R8 and R3 soldered with resistor and R5 is unsoldered for UART activation) steppers on UART.
But now it is working well.

I set all drivers in aurt mode first then boot the board, No matter i did it was giving this error ''Unable to read tmc uart 'stepper_y' register IFCNT.
I removed tmc2209 stepper_y from config then it was giving same error for stepper_x

Later, i added every steppers one by one without uart settings, then i set each uart settings one by one with rebooting mcu at each additions. Then it is fixed.
Dont know if it fix your problem too.

Just be carefull about not putting stepper drivers while board is powered on.

closed 2021-02-14 by KevinOConnor

cmroche, 2021-03-22:
Hey, just want to add that I had the same issue here. Did a firmware update to latest on master about an hour ago, updated firmware and then i would get `Unable to read tmc uart 'stepper_x' register IFCNT.`

I followed @whiterabbit62 's instructions, commented out the `[tmc2209 stepper_x]` section of my config, restarted, homed X, then added the section back any now everything is working once again.