Klipper3d/klipper #3903: M104 S170 is ignored at start of print

kubroid, 2021-02-08:
Hi guys,
I have some strange bug - extruder does not start to heat up by M104 S170 command in gcode file.
If I issue this command(s) in the console - no problem at all.

Here is my test program:
```
M104 S170 ; set extruder temp for bed leveling
M140 S80 ; set bed temp
M109 R170 ; wait for bed leveling temp
M190 S80 ; wait for bed temp

M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
```

After start, only bed is heating until reach 80C, and only then extruder starts to heat up.

Also, I've found another possible issue - it's impossible to cancel the program during bed heating up.
here is console output:
```
13:06:18 File opened:test.gcode Size:198
13:06:18 File selected
13:06:23 $ CANCEL_PRINT
13:10:22 // action:paused
13:10:22 // action:cancel
13:10:22 // action:paused
13:10:22 // action:cancel

```
[klippy (3).log](https://github.com/KevinOConnor/klipper/files/5943814/klippy.3.log)
 -->

kubroid, 2021-02-08:
Log with M112
[klippy.log](https://github.com/KevinOConnor/klipper/files/5944332/klippy.log)

schnello, 2021-02-10:
Hello

Your lines
```
M104 S170 ; set extruder temp for bed leveling
M140 S80 ; set bed temp
M109 R170 ; wait for bed leveling temp
M190 S80 ; wait for bed temp

M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
```

M104 S170 set the Hotend Temp ... OK
M140 S80 ; set bed temp ... OK

M109 **R**170 ; wait for bed leveling temp
R is not valid in Klipper.... so Klipper change the Temp to 0

kubroid, 2021-02-10:
@schnello , Thank you so much!
My fault.

closed 2021-02-10 by kubroid