Klipper3d/klipper #3907: Stutter at arcs and "Timer too close"

PerLangkjaer, 2021-02-09:
Hello
I have had some issues with stutter when during arcs.
I have now begun trying to fix it instead of just turning the speed down.
I got that error with the Arc Welder plug-in.
Is it possible that someone have an idea what to to?

The two .txt files is the gcode im running for testing this.

[CL1 v1.txt](https://github.com/KevinOConnor/klipper/files/5949232/CL1.v1.txt)
[CL1_v1.aw.txt](https://github.com/KevinOConnor/klipper/files/5949233/CL1_v1.aw.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5949234/klippy.log)

th33xitus, 2021-02-09:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

PerLangkjaer, 2021-02-09:
Sorry to have taken up space here

closed 2021-02-09 by PerLangkjaer