Klipper3d/klipper #3908: M204 with Super Slicer not working

markwinger, 2021-02-09:
I have been trying to use super slicer the last few days (was using cura).  I got things setup prettey well and printeed a test printer.  I did a few other things then reprinted the same object without making any changes to the slicer.  When printing the second time it had taken about twice as much time with 30% to go.  I could tell my listening to the printer that acceleration was much slower.  I noticed the following error on the console:
Invalid M204 command "M204 P3500 ; adjust acceleration"

The klipper doumentation of gcode says that if the "Note: If S is not specified and both P and T are specified, then the acceleration is set to the minimum of P and T. If only one of P or T is specified, the command has no effect."
I typed M204 P3500 T3500 in the console and the speed jumped way up. 
Looking an the marlin documention on gcode it says "Legacy M204 S<accel> is deprecated. Use separate paremeters M204 P<accel> T<accel> instead.".  Usage says:  "M204 [P<accel>] [R<accel>] [S<accel>] [T<accel>]"

This implies to me that M204 S<acc> changed both print and travel acceleration.  To get the same behavior use M204 P<acc> T<acc>.  But it does not say that you must include both P<> and T<> for the command to be valid.

I suspect that since super slicer is a fork of prusa slicer and slic3r, the they my have the same problem.

I think klipper ignoring the M204 P<>  command is a mistake in kilpper.  Either way it is in conflict with some slicers so I believe this is a but in klipper.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5952094/klippy.log)

jakep82, 2021-02-09:
This behavior is documented.  https://www.klipper3d.org/G-Codes.html

`Set acceleration: M204 S<value> OR M204 P<value> T<value>`

`Note: If S is not specified and both P and T are specified, then the acceleration is set to the minimum of P and T. If only one of P or T is specified, the command has no effect.`

markwinger, 2021-02-09:
You quoted the same thing I quoted from the klipper g-code documentaion.  Klipper IS working as designed based on that.  But Marlin docs say this is wrong.  If klippers interpretation is correct, usage would be defined more like:

M204 [P<accl> T<accl>]  [S<accl>] [R<accel]

This would say P and T must be used together.

But marlin spec says:

Usage: M204 [P<accel>] [R<accel>] [S<accel>] [T<accel>]

Which specifically says all options are optional, P and T are not required to be supplied together.

These two interpretations are not compatible.....so slicers that follow the specified usage as super slicer does, will not work correcly with klipper...
What I am saying is that klippers interpretation will not work with some slicers unless the slicers all change to follow klippers interpretation of marlin.

Is there is some other spec was used that Marlin, super slicer, prusa slicer,... should have used that implies that they interpreted this incorrectly?
Am I missing something or looking at this all wrong???

jakep82, 2021-02-09:
Klipper is not Marlin.  This is the documented and intended behavior for Klipper.

KevinOConnor, 2021-02-09:
Ah, thinking about this further, I'm going to remove the "invalid" label.  I understand your report, but I do not agree this is a defect in Klipper.  It is not a goal of Klipper to be compatible with Marlin.  The goal of Klipper's g-code support is to be compatible with slicers in their standard configurations - slicer based acceleration control is not a requirement to use a slicer (nor is it particularly useful in my opinion).

-Kevin

Sineos, 2021-02-09:
See https://github.com/prusa3d/PrusaSlicer/issues/1089#issuecomment-759375802

Hywelmartin, 2021-02-09:
superslicer klipper as G-code flavor.... for other Slic3r derivatives I use RepRap/sprinter
[image]

markwinger, 2021-02-09:
I know its not marlin.  But in the slicers page of your docs ( https://www.klipper3d.org/Slicers.html ) it says Many slicers have an option to configure the “G-Code flavor”. The default is frequently “Marlin” and that works well with Klipper.  

Thanks Hywelmartin.  I had not seen where firmware flavor was setup.  I changed it to klipper and it generates M204 S<accel> where as marlin flavor generates M204 P<accel>.    

Sorry for my mistake.

closed 2021-03-10 by github-actions[bot]