Klipper3d/klipper #3910: MCU 'mcu' shutdown: Timer too close

leadgtr7, 2021-02-10:
MCU 'mcu' shutdown: Timer too close
This is generally indicative of an intermittent
communication failure between micro-controller and host.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

This is second time I have seen this issue. I have read through the times this has been brought up in github issues and the general consensus is to update to latest. I looked at the log file and all my temps were within range. At the time of print start I was on the latest available versions

This happened 5 hours into a 5.5hr print (gcode file attached). While I understand that I am to use the discord I don't readily see where I should post this issue.
[KlipperMCUfailure log_2-9-2021.log](https://github.com/KevinOConnor/klipper/files/5955000/KlipperMCUfailure.log_2-9-2021.log)

[Voron-V1-ABS_MainBody_5h34m.zip](https://github.com/KevinOConnor/klipper/files/5955031/Voron-V1-ABS_MainBody_5h34m.zip)

I am running klipper, moonraker, and mainsail at the following versions
0.9.1-216
0.3.0-4
1.1.0

Printer is Voron V1.8 with SKR v1.3 board and TMC2209 drivers all around

leadgtr7, 2021-02-10:
This happened again, This time when I started the a file after a 20 min heatsoak
[KlipperMCUfailure log_2-10-2021.log](https://github.com/KevinOConnor/klipper/files/5958585/KlipperMCUfailure.log_2-10-2021.log)

BlackStump, 2021-02-10:
[FAQ](https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors)

leadgtr7, 2021-02-10:
@BlackStump thanks for pointing me to that FAQ page. So you are saying that FAQ section addresses the Timer too Close condition?

Is this error not indicative of something? It didn't seem like a random error as is called out in the FAQ.

leadgtr7, 2021-02-12:
[KlipperMCUfailure log_2-11-2021.log](https://github.com/KevinOConnor/klipper/files/5969107/KlipperMCUfailure.log_2-11-2021.log)

Another failure with same issue. I have checked all connections and replaced the usb cable between SKR v1.3 and Rpi cable yesterday after last error.

Is there something in these logs that would indicate what is going on here? I am kinda pulling my hair out on this

KevinOConnor, 2021-02-12:
Your first log doesn't actually contain an error, but the second two do.  At the low-level, both logs show that the heater update thread did not get any CPU time for an extended period (about 1 second in the first log, and about 500ms in the second log).  This is an indication of a problem at the operating system level on the host computer.

Unfortunately, the logs don't indicate the source of that problem.  Typical reasons are bad sdcard, some process on the host using excessive amounts of memory, voltage problems on the host, etc.  If you are using a webcam connected to the host, there have also been reports that some of them can cause problems.

If you're familiar with Linux, you can try to review the system logs at the time of the event.  Those logs may provide some information on what was occurring at the OS level during the event.

-Kevin

leadgtr7, 2021-02-12:
Thanks so much for everyone trying to help me on this!
I have ordered a dedicated power supply for the pi instead of a buck converter
I will also replace the SD card with a sandisk class10 32GB one
In addition i have the webcam set to 1080p 30 fps and i will reduce this dramatically.

It sounds like I have a solution going forward. Thanks so much for the help on this.

closed 2021-02-12 by leadgtr7