Klipper3d/klipper #3913: Manual stepper questions

webers89, 2021-02-10:
1. I went through documentation about manual stepper, but couldn't find a way to switch from absolute to relative coordinates.
Is it possible to use relative coordinates for manual stepper? 

2. Is there a way to synchronise two manual steppers similar like it is possible to stepper_x and stepper_x1 ?
I tried to make g code macros to move both manual steppers but second stepper starts to move only after the first one has stopped. I made macros with manual stepper and X-axis stepper move at the same time then it worked, but not for two manual steppers at the same time.

jakep82, 2021-02-10:
https://www.klipper3d.org/Contact.html#i-have-a-question-about-klipper

KevinOConnor, 2021-02-11:
In short, however, 1) No (though maybe through the use of elaborate macros), and 2) no.  Neither is currently supported.

-Kevin

closed 2021-02-14 by github-actions[bot]