Klipper3d/klipper #3914: Gcode created by Simplify3D ERROR

pgh268400, 2021-02-10:
Hello. Good morning. I installed Klipper this time. And I created Gcode with Simplify 3D.

After uploading to Octoprint and heating is complete, the printout is not printed.

English is not my first language. Please understand my bad skills. Thank you.

**Error Code :**
 Recv: !! Error on 'M104 S{material_print_temperature_layer_0}': unable to parse {
Changing monitoring state from "Printing from SD" to "Cancelling"
Send: M108
Send: M25
Recv: // Unknown command:"M108"

**Gcode :** 
G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 S{material_print_temperature_layer_0}
M140 S{material_bed_temperature_layer_0}
M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature_layer_0}
G28 X0 Y0
G28 Z0
M420 S1
G1 Z5 F3000
G92 E0.0
G1 Y-3.0 Z2 F1000.0
G1 X60.0 E9.0 F700.0
G1 X100.0 E21.5 F700.0
G92 E0
G1 X120 E3 Z0.0 F800
G1 X160 F500
G92 E0.0
M900 K40

jakep82, 2021-02-10:
Your gcode is faulty.  You've used variables that weren't postprocessed properly by S3D.

```
M104 S{material_print_temperature_layer_0}
M140 S{material_bed_temperature_layer_0}
M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature_layer_0}
```

All of these variables need to be replaced by numbers during processing in S3D.  They are not valid gcodes, and no firmware could properly parse them.

pgh268400, 2021-02-11:
Does that mean that the S3D I use caused an error during the Gcode conversion process?
In Marlin it reads correctly, but I don't know why.

pgh268400, 2021-02-11:
> Your gcode is faulty. You've used variables that weren't postprocessed properly by S3D.
>
> ```
> M104 S{material_print_temperature_layer_0}
> M140 S{material_bed_temperature_layer_0}
> M190 [...]

========================================
Does that mean the S3D I'm using caused an error during the Gcode conversion process?
In Marlin it reads correctly, but I don't know why.

schnello, 2021-02-11:
material_print_temperature_layer_0 is a variable in Cura. So i guess if you say in Marlin was all fine... you create the Gcode with Cura.

pgh268400, 2021-02-11:
thank you for the reply.
But I actually generated the above Gcode in Simplify3D and printed it in Marlin. Perhaps Marlin ignores the error.
Once I modified the Gcode above and printed it with Klipper, I succeeded, but the extrusion did not work properly. :(
The amount of extrusion is too small.

There seems to be a problem with rotation_distance. Could you please check if my calculation method is correct?

I set the rotation_distance of the extruder to 34.40.

rotation_distance = <full_steps_per_rotation> * <microsteps> / <steps_per_mm>

My Marlin Steps Per Unit : 
/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2...]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 800, 93 }

I use 1.8 degree steppers, tmc2209(microsteps may be 16?)

thefore extruder rotation_distance = 
200 * 16 / 93 = 34.4086....

closed 2021-02-14 by KevinOConnor