Klipper3d/klipper #3922: Bigtreetech GTR KTEM SPI BUS not implemented

jatson, 2021-02-12:
Hi,

I was trying to use a K-type thermocouple with a Bigtreetech GTR but I'm getting the following error:
Transition to shutdown state: MAX31855 : Open Circuit
MCU 'mcu' shutdown: Thermocouple ADC out of range

If I'm using the thermocouple with AD8495 connected to the T0 connector it is working fine, so I assume the sensor and the wiring is OK.
I tried to check inside the code that maybe I'm using wrong SPI_BUS config and I found out that the correct BUS is not seems to be implemented:
according to the schematics the pins for this shall be the following:
[image]

but as I can understand only these are implemented:
[image]

I've attached the config I'm using currently (this is a fresh new printer, so it might be uncomplete) and also the log is attached.

Please excuse me if I'm not wrong, I'm new with klipper.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5971773/klippy.log)

[printer_config.zip](https://github.com/KevinOConnor/klipper/files/5971775/printer_config.zip)

KevinOConnor, 2021-02-12:
I added in a new definition for a `sp2b` bus (commit fc783f9b).  Can you pull the latest code and retry (see the FAQ for upgrade instructions).

-Kevin

jatson, 2021-02-13:
working perfectly!
Thanks for the very quick solution.

closed 2021-03-06 by jatson