Klipper3d/klipper #3923: Raspberry isn't finding the USB Device after reflashing the SKR Board

nicedevil007, 2021-02-12:
hello,

today we installed fluiddpi with everything preinstalled for starting with klipper.
we followed the guide on the klipper documentation to get the klipper.bin for the skr 1.4 turbo board.

So we tryed the following settings:

[image]

and we also tryed those settings:

[image]

both didn't work. We also enabled serial on the raspi-config.

lsusb won't show the connected device. The device was working before flashing with octoprint and marlin, so I guess it has to be a problem with the firmware?

nicedevil007, 2021-02-13:
Fixed, broken USB cable

closed 2021-02-13 by nicedevil007

theopensourcerer, 2021-02-13:
You do not provide a klippy.log file so we are unable to see your configuration and/or any errors that klipper may be reporting. 

Have you checked your configuration is using the correct device? i.e.:

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

Also, in the config example for the SKR 1.4 it says "This file contains common pin mappings for the BIGTREETECH SKR V1.4
# board. To use this config, the firmware should be compiled for the LPC1768 or LPC1769(Turbo)."

Did you try using the 1768?