Klipper3d/klipper #3924: Unknown command: queue_digital_out

wuast94, 2021-02-12:
I have updated my Klipper today, after changing from step_distance to rotation_distance and adding microsteps in config i getting errors on start up.
 Unknown command: queue_digital_out
 
 i update the firmware by copy the klipper.bin to sd card. 
 
 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5974467/klippy.log)

pls ask if you need more information :) im using an bigtreetech 1.4 turbo

wuast94, 2021-02-12:
```
I found a bug in Klipper

Klipper is an open-source project and we appreciate when collaborators provide bug reports.

There is important information that will be needed in order to fix a bug. Please follow these steps:

1.    Be sure the bug is in the Klipper software. If you are thinking “there is a problem, I can’t figure out why, and therefore it is a Klipper bug”, then do not open a github issue. In that case, someone interested and able will need to first research and diagnose the root cause of the problem.
``` 
its a klipper related bug

`2. Make sure you are running unmodified code from https://github.com/KevinOConnor/klipper. If the code has been modified or is obtained from another source, then you will need to reproduce the problem on the unmodified code from https://github.com/KevinOConnor/klipper prior to reporting an issue.`

its unmodified

` 3.   If possible, run an M112 command in the OctoPrint terminal window immediately after the undesirable event occurs. This causes Klipper to go into a “shutdown state” and it will cause additional debugging information to be written to the log file.`
if done the M112 command

`4.   Obtain the Klipper log file from the event. The log file has been engineered to answer common questions the Klipper developers have about the software and its environment (software version, hardware type, configuration, event timing, and hundreds of other questions).`

ive added the klippy.log file

what did i miss that this is an invalid issue? the only thing is that i removed the doubled code from log to make it easier for you, here is now the full log: 

[klippy.log](https://github.com/KevinOConnor/klipper/files/5974524/klippy.log)

[klippy.log](https://github.com/KevinOConnor/klipper/files/5974511/klippy.log)

raymondh2, 2021-02-12:
This error usually means you need to update the firmware on your MCU board.

wuast94, 2021-02-12:
Git version: 'v0.9.1-226-g5db0b3d4'
 Firmware version: v0.9.1-226-g5db0b3d4
 
 seems fine to me

Gilabite, 2021-02-12:
Did you upgrade the pi as mcu firmware too, not just SKR?

wuast94, 2021-02-12:
im just stupid, ive just copy the klippy.bin but the naming must be firmware.bin

now it seems to work :) i was confused becouse of the "Firmware version: v0.9.1-226-g5db0b3d4" line, was thinking flashing was succesfull but it wasnt

maybe a thing were the output could be improved but yeah now its working :) 

and would be realy nice to not get just an invalid tag, a short answere what is missing would be way more helpfull :)

closed 2021-02-12 by wuast94