Klipper3d/klipper #3932: PWM_MAX FW constant missing for LPC1769

zellneralex, 2021-02-14:
I tried to setup the following:
[fan]
pin = P2.1
kick_start_time = 0.5
off_below = 0.10
cycle_time = 0.00005
hardware_pwm = true

Pin 2.1 is a HW PWM Pin. I get a fail that says fw constant missing.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5978178/klippy.log)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5978179/klippy.log)

Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 151, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 581, in _connect
    self._send_config(None)
  File "/home/pi/klipper/klippy/mcu.py", line 520, in _send_config
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 203, in _build_config
    self._pwm_max = self._mcu.get_constant_float("PWM_MAX")
  File "/home/pi/klipper/klippy/mcu.py", line 702, in get_constant_float
    return self._serial.get_msgparser().get_constant_float(name)
  File "/home/pi/klipper/klippy/msgproto.py", line 404, in get_constant_float
    return self.get_constant(name, default, parser=float)
  File "/home/pi/klipper/klippy/msgproto.py", line 396, in get_constant
    raise error("Firmware constant '%s' not found" % (name,))
error: Firmware constant 'PWM_MAX' not found
Build file /home/pi/klipper/klippy/../.config(1029): Sun Feb 14 18:37:56 2021
========= Last MCU build config =========
CONFIG_LOW_LEVEL_OPTIONS=y
# CONFIG_MACH_AVR is not set
# CONFIG_MACH_ATSAM is not set
# CONFIG_MACH_ATSAMD is not set
CONFIG_MACH_LPC176X=y
# CONFIG_MACH_STM32 is not set
# CONFIG_MACH_PRU is not set
# CONFIG_MACH_LINUX is not set
# CONFIG_MACH_SIMU is not set
CONFIG_STEP_DELAY=2
CONFIG_BOARD_DIRECTORY="lpc176x"
CONFIG_MCU="lpc1769"
CONFIG_CLOCK_FREQ=120000000
CONFIG_USBSERIAL=y
CONFIG_FLASH_START=0x4000
CONFIG_FLASH_SIZE=0x80000
CONFIG_RAM_START=0x10000000
CONFIG_RAM_SIZE=0x7fe0
CONFIG_STACK_SIZE=512
CONFIG_LPC_SELECT=y
# CONFIG_MACH_LPC1768 is not set
CONFIG_MACH_LPC1769=y
CONFIG_SMOOTHIEWARE_BOOTLOADER=y
CONFIG_USB_VENDOR_ID=0x1d50
CONFIG_USB_DEVICE_ID=0x614e
CONFIG_USB_SERIAL_NUMBER_CHIPID=y
CONFIG_USB_SERIAL_NUMBER="12345"

FHeilmann, 2021-02-14:
LPC176x doesn't support hardware pwm, remove 
`hardware_pwm = true`
from your config and everything should start working.

zellneralex, 2021-02-15:
It does, will need to connect a scope to see if that is running the needed requested 20 KHz PWM signal for the fan with SW PWM.

zellneralex, 2021-02-16:
If possible a better Error message would be helpful.

closed 2021-03-17 by github-actions[bot]