Klipper3d/klipper #3948: Neopixel update M109

xbladen, 2021-02-16:
Hi,
i'm struggeling with neopixel connected to temperture.
Tried the M105 replacement macro, which works itself, if the slicer would use it.
During heatup M109 is used, and actual temperature is reported back, but no neopixel update is possible or M105 command is executable to update neopixel.
Normally i would wish to connect neopixel to the temperature feedback, like the M105 way, a macro that updates
neopixel, if neopixel output has changed (not every feedback message), but not using polling G-Codes.
I guess this is not supported yet ?
Neopixel examples are quite rare - am i missing something ?
Maybe it could be possible to connect macro's to functions - like temperature feedback, endstop reached, entering idle ...

I provided a log-file just pro forma ...
[klippy.log](https://github.com/KevinOConnor/klipper/files/5992295/klippy.log)

xbladen, 2021-02-17:
Let me think this idea a little bit further - please not that i'm not socialized with python :-)
extend printer.cfg by function_macro like :
[function_macro endstop_open]
    SET_LED LED=some_led RED=1.0 GREEN=0.0 BLUE=0.0
[function_macro endstop_close]
    SET_LED LED=some_led RED=0.0 GREEN=1.0 BLUE=0.0

extend configfile.py by getfunctionmacro (like getint), return the macro itself or a pointer to macro if found in configfile, else none.

create a function execute_function_macro(macro), that (if macro is not none) executes the macro like gcode_macro, maxbe some parts of gcode_macro could be used.

in the interesting fuctions use it like:
self.function_macro_endstop_x_change = getfunctionmacro(configfile, "endstop_x_change")
self.function_macro_endstop_x_open = getfunctionmacro(configfile, "endstop_x_open")
self.function_macro_endstop_x_close = getfunctionmacro(configfile, "endstop_x_close")

self.endstop_x.state = newstate(enstop_x)
if self.endstop_x.state!=self.endstop_x.laststate:
    execute_functionmacro(self.function_macro_endstop_x_change)
    if self.endstop_x.state==0:
        execute_functionmacro(self.function_macro_endstop_x_open)
    else:
        execute_functionmacro(self.function_macro_endstop_x_close)
self.endstop_x.laststate = self.endstop_x.state

This would be useable in any function of interest 
- report temperature
- state_change (enter printing, exit printing)
- endstop changes
- ...
You could extend klipper by many funtions and switch pins, execute GCodes, etc. just by defining macros in configfile.

Could be much easier than gcode_macro, because you don't need to insert new gcodes or new function calls, just use existing ones in klippy. Nothing to be supported in klipper firmware itself. Unfortunately i don't know if "getfunctionmacro" could return a complete macro - maybe just plain text.
Maybe you need a list of macros - add a macro to the list if getfunctionmacro is called and return the pointer.
I think you don't want to parse configfile during execute_functionmacro.

xbladen, 2021-02-17:
Hmmm spaces at the beginning of a line are deleted - so how do i insert tabs here ?

xbladen, 2021-02-17:
self.endstop_x.state = newstate(enstop_x)
if self.endstop_x.state!=self.endstop_x.laststate:
	execute_functionmacro(self.function_macro_endstop_x_change)
	if self.endstop_x.state==0:
		execute_functionmacro(self.function_macro_endstop_x_open)
	else:
		execute_functionmacro(self.function_macro_endstop_x_close)
self.endstop_x.laststate = self.endstop_x.state

xbladen, 2021-02-17:
Hmmpf tabs are deleted, too ...

xbladen, 2021-02-17:
_self.endstop_x.state = newstate(enstop_x)
_if self.endstop_x.state!=self.endstop_x.laststate:
_	execute_functionmacro(self.function_macro_endstop_x_change)
_	if self.endstop_x.state==0:
_		execute_functionmacro(self.function_macro_endstop_x_open)
_	else:
_		execute_functionmacro(self.function_macro_endstop_x_close)
_self.endstop_x.laststate = self.endstop_x.state

xbladen, 2021-02-17:
Arrrgh: tabs are deleted everywhre

Turge08, 2021-02-18:
@xbladen I like where this is headed. Were you able to confirm this works at all?

Sineos, 2021-02-18:
https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks

xbladen, 2021-02-18:
First at all,
this was an idea, so not working stuff.
I have mixed gcode_macro with some other stuff and have now (hopefully) function_macro- printer objects generated from config file.
config file is like described:
'''
[function_macro endstop_open]
macro:
  SET_LED LED=some_led RED=1.0 GREEN=0.0 BLUE=0.0
'''
and can contain macros like gcode macros.
The function_macro object has
- name
- script (contecnt of macro)
- run_macro_as_command (this should run gcode-commands, just like a gcode macro)

What is missing is : how to get a function_macro from printer with specific name
When this works, it would be possible to call the macro.
Then this would need to be inserted in the desired functions, but that would be copy & paste.

Maybe i will find something at pin-naming, because that uses pins by name, but maybe this is just done in firmware.

There is a caveeat, to insert macros in functions will lead to maybe unpredictable timing problems.
On the one hand the macros are scanned for commands and executed line by line - consumes runtime.
On the other hand commands like for neopixel may comsume runtime by itself.
So this should not be used by excess, but if you just need a pin to be switched when something happens, or you are waiting for a temperature to be reached, you can do such neopixel-stuff or switch a pin.

Onother way could be, that all theese functions raise condition changes to a central component, that disppatches the right function_macro.

Unfortunately i'm no python-native.

xbladen, 2021-02-18:
A short update - hopefully seems to work.
I have a function_macro-printer-object, that builds a list of macros together with a list of names.
Can execute macro by name.
macros can contain gcode-commands (not scripts), but can contain calls to defined gcode macros, that contain scripts.
So i guess i can switch directly pins, execute commands or execute gcodes with embedded scripts.
But i don't know, if gcodes are executed, when other gcodes still are running (like M109), or if they go to a fifo.
So i need to set it up in real life ...
The restriction to not execute scripts directly is from gcode.run_script_from_command, that executes gcode-commands, the script-execution is done when executing the gcode-macro, but as long as i can call a gcode-macro, scripting can be done there.

xbladen, 2021-02-19:
looks promising
everything seems to work

xbladen, 2021-02-20:
Code works and is committed, lasts in pull_request.
If someone wants to try :
https://github.com/xbladen/klipper

xbladen, 2021-02-20:
Issues are little bit like talking to yourself, aren't they ?

axg20202, 2021-02-25:
I just found this thread. Interested in this functionality as, coming from Marlin, it was very useful to have neopixels gradually change colour automatically with changing temperature without having to resort to Gcode commands to do it. It was at firmware level. Gradual colour fading from blue through several shades to red during heatup, then switch to all white at print start to light the job. Not sure what you were developing above but would be willing to try some stuff out if you can give me some pointers please. Where do I find the code you mentioned?

xbladen, 2021-02-25:
@avg20202
my part can be found at https://github.com/xbladen/klipper
what is a fork of klipper, but i had a discussion with kevin o'conner and this seems not to work stable the way it was build, because it executes gcode-commands, where it maybe is not allowed. gcode-commands need to be an atomic action. neopixel can be updated at any time, but it needs a separate entry point beside executing gcode. Will try to move this direction on the weekend.
In fact i face some crashes now i didn't have before, at the moment the printer does nothing but neopixel and crashes after some hours.
What i wanted was to update neopixel during heatup. The solution so far tried to use execution gcodes and by that using gcode-macro's. I will try to copy macro stuff, to keep defining the function in configfile and provide two commands for that :
SET_LED and SET_PIN. So you cannot do everything gcode_macro can do, but update neopixel any time and update pins-output anytime - but you should not use pins already used for other stuff.

axg20202, 2021-02-25:
Thanks. Yes, independent of gcode is the way to go for this. My MCU has a dedicated pin for neopixel (it is actually the servo pin, which is not used elsewhere) so no risk of jumbling up pins for me. I wonder if looking into how it was implemented in Marlin would give any clues?

xbladen, 2021-02-25:
@avg20202
Think about how Klipper works, klippy calculates the stepper steps and sends this to klipper to put it into a timer-fifo. When the timer expires, the step is done by klipper.
klippy get a set_led and calculates many pin-changes for neopixel-pin and sends them to the timer-fifo to be played back.
So it is not just a task, if neopixel-pin is a pin only used for this, it is also a task of not corrupting the timer-playback-fifo.
The second idea could be to implement it fixed in firmware, like marlin. But what if someone wants to use it another way, other colors, light led's one by one on rising temperature ?
I like the idea of scripting this and let everybody use its own script. But both ways are reasonable.
My approach works (sometimes) and i would like to go this way. Looking into marlin could give ideas of how to build the script and maybe what is missing - what can marlin do, that the scricpt cannot ?
On my side i would like to switch of the lights after a timeout, when nothing happens, so i would like to have a printerstate idle, that does not exist till now - another task.

axg20202, 2021-02-25:
Re lighting LEDs sequentially, Marlin has this too. I don't know if it helps at all, but this is how printer event LEDs are implemented in the Marlin source code: 

#include "printer_event_leds.h"

PrinterEventLEDs printerEventLEDs;

#if HAS_LEDS_OFF_FLAG
  bool PrinterEventLEDs::leds_off_after_print; // = false
#endif

#if HAS_TEMP_HOTEND || HAS_HEATED_BED

  uint8_t PrinterEventLEDs::old_intensity = 0;

  inline uint8_t pel_intensity(const float &start, const float &current, const float &target) {
    if (uint16_t(start) == uint16_t(target)) return 255;
    return (uint8_t)map(constrain(current, start, target), start, target, 0.f, 255.f);
  }

  inline void pel_set_rgb(const uint8_t r, const uint8_t g, const uint8_t b) {
    leds.set_color(
      MakeLEDColor(r, g, b, 0, neo.brightness())
      #if ENABLED(NEOPIXEL_IS_SEQUENTIAL)
        , true
      #endif
    );
  }

#endif

#if HAS_TEMP_HOTEND

  void PrinterEventLEDs::onHotendHeating(const float &start, const float &current, const float &target) {
    const uint8_t blue = pel_intensity(start, current, target);
    if (blue != old_intensity) {
      old_intensity = blue;
      pel_set_rgb(255, 0, 255 - blue);
    }
  }

#endif

#if HAS_HEATED_BED

  void PrinterEventLEDs::onBedHeating(const float &start, const float &current, const float &target) {
    const uint8_t red = pel_intensity(start, current, target);
    if (red != old_intensity) {
      old_intensity = red;
      pel_set_rgb(red, 0, 255);
    }
  }
#endif

#endif // PRINTER_EVENT_LEDS

closed 2021-03-26 by github-actions[bot]