Klipper3d/klipper #3949: using [temperature_sensors] as sensors in [heater] or [temperature_fans]

ronald5555, 2021-02-17:
Why is it not possible to use a temperature sensor defined in an [temperature_sensors] section as a sensor for a  [heater] or [temperature_fans].
My concern is that I need to use the same temperature_sensor (chamber temperature) on time as a sensor for the chamber heater and one time for the exhaust fan (during cooling phase).
If I use the same sensor in these 2 sections I get the error "sensor used multiple times" 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5994475/klippy.log)

theopensourcerer, 2021-02-17:
If I understand correctly I think you shouldn't declare your device as a [temperature_sensor] as well as a [temperature_fan]. I use a DS18B20 connected to my RPI to control my enclosure fans and the configuration for my RPi mcu is simply this - I do not define the device as a sensor, only as the sensor for my fan:

```
[mcu host]
serial: /tmp/klipper_host_mcu

[temperature_fan enclosure]
pin: host:gpio17
sensor_type: DS18B20
serial_no: 28-0315a1468bff
sensor_mcu: host
ds18_report_time: 1.5
min_temp: 0
max_temp: 60
gcode_id: DSB1
target_temp: 30.0
control: pid
pid_Kp: 40
pid_Ki: 0.2
pid_Kd: 0.1
```

Does that help? Sorry if I misunderstood.

ronald5555, 2021-02-17:
Hello @theopensourcerer .
Your way is the way I do it today. But I want to add a chamber heater that uses the same sensor as the exhaust fan.
This way I can heat the chamber before and during the print process and use the exhaust fan after printing until chamber is at room temp.

`[temperature_fan controller_fan2]
pin: FAN4
sensor_type: ATC Semitec 104GT-2 
sensor_pin: TEMP_3 

[heater_generic chamber_heater]
heater_pin: HEAT_4
sensor_type: ATC Semitec 104GT-2
sensor_pin: TEMP_3
`

As you see in the above code, the same sensor is used twice and that is the problem !!

theopensourcerer, 2021-02-17:
I'm not sure that is a supported configuration... I would think your best bet initially would be to discuss this on the discord channel first and/or then create a pull request for this feature if it isn't supported already: 

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#discord

closed 2021-02-18 by ronald5555