Klipper3d/klipper #3950: [FR] - Possibility to use several endstops on a single manual stepper

EtteGit, 2021-02-17:
Hello,

I'm developing a MMU system and it would perform much better if one could use different endstops with a single manual stepper (as far as I know this is not possible yet).

The idea would be to be able to either : change the Pin for the endstop using a gcode (so you first change the pin and then you "home" your manual stepper) or to simply define several endstops in the cfg for the manual stepper and then add a variable to define which one to use when doing a homing.

In a parfect world such an endstop would also be a filament_sensor_switch (not while homing the manual stepper of course), but this would just be cherry on the cake.

Is such a thing possible?

Thanks

KevinOConnor, 2021-02-18:
FYI, we choose not to use github to make feature requests.  (See docs/Contact.md )  However, if you wish to add something like this, I'm sure it's possible.

-Kevin

EtteGit, 2021-02-18:
Oh sorry, my bad!

I'll take a look on the discord.

Thanks for the reply

Ette

closed 2021-02-18 by EtteGit