Klipper3d/klipper #3953: Using GCodePrintr App to control 3D printer with Klipper

dietzm, 2021-02-17:
Android App GCodePrintr works fine with Marlin, Repetier and other 3D printing firmwares by talking gcode to the printer board through USB (Bluetooth / Wifi can be used as well, but is less common)
A typical setup is that GCodePrintr is installed on a phone or tablet which uses a USB OTG adapter to connect to the USB port of the printer board.

My understanding is that Klipper does not act as a USB host , but uses USB only to talk to the printer board, right ? 
Since the GCodePrintr App is running on a different device (phone/table) it can't use the virtual serial device file offered by Klipper directly. 

Is it possible to let Klipper act as a USB host and receive gcode cmds through USB ? 
Is there an alternative solution ?

Thanks
Mathias

jakep82, 2021-02-18:
https://www.klipper3d.org/Contact.html#i-have-a-question-about-klipper

dietzm, 2021-02-18:
Ok, my fault. Let me turn this into a feature request: 
Klipper needs to act as a USB host device to so that other 3rd party host applications (like GCodePrintr) can send gcode cmds to it over USB.

closed 2021-02-22 by KevinOConnor