Klipper3d/klipper #3954: Error in Octopi terminal when starting Klipper for the first time

cedorland, 2021-02-17:
I just installed Klipper for the first time, and followed all of the installation directions.
When I get to the end of the installation instructions:
"It is not unusual to have configuration errors during the initial setup - update the printer config file and issue “restart” until “status” reports the printer is ready."
I have been sitting at my octoprint terminal using the restart and status commands for over an hour.
Restart does this:

```
Send: restart
Recv: ok
Recv: // Klipper state: Disconnect
```

and then nothing.
Status gives me this:

```
Send: status
Recv: // Internal error during connect: Choice 'item' for option '{'disabled': <class 'extras.display.menu.MenuDisabled'>, 'input': <class 'extras.display.menu.MenuInput'>, 'list': <class 'extras.display.menu.MenuList'>, 'command': <class 'extras.display.menu.MenuCommand'>, 'vsdlist': <class 'extras.display.menu.MenuVSDList'>}' is not a valid choice
Recv: //
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: // Klipper state: Not ready
Recv: !! Internal error during connect: Choice 'item' for option '{'disabled': <class 'extras.display.menu.MenuDisabled'>, 'input': <class 'extras.display.menu.MenuInput'>, 'list': <class 'extras.display.menu.MenuList'>, 'command': <class 'extras.display.menu.MenuCommand'>, 'vsdlist': <class 'extras.display.menu.MenuVSDList'>}' is not a valid choice
Recv: ok
```
After reading that log file, I deleted all references to `[menu __(name)]` in the printer.cfg file, and after restarting klipper, I got the following from OctoPrint:

```
Send: status
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: // Klipper state: Not ready
Recv: !! Printer is not ready
Recv: ok
Send: status
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: // Klipper state: Not ready
Recv: !! Unable to connect
Recv: ok
```
 
After doing this, my printer's LCD screen is blank. I've restarted the printer, same thing. I also get the following error in bash:
```
pi@octopi:~/klipper $ ls /dev/serial/by-id/*
ls: cannot access '/dev/serial/by-id/*': No such file or directory
```

It looks like I'll have to flash my printer again at this point. I was really excited to use Klipper, but I'm going to give up for now and reinstall OctoPrint from scratch. Did I do something wrong? I followed all of the instructions to the dot, or so I thought.

Here's the latest [klippy.log](https://github.com/KevinOConnor/klipper/files/5999293/klippy.log), although it just tells you that it can't find the printer because it somehow deleted the printer's firmware. The address it's looking at *was* correct, until I deleted all references to `[menu __(name)]` in the cfg file and restarted Klipper.

jakep82, 2021-02-18:
https://www.klipper3d.org/Contact.html#help-it-doesnt-work

Please try the Discord for help with initial setup.

cedorland, 2021-02-18:
Okay, now I can't access my printer via the Arduino IDE, so I can't even reflash it. Seems that Klipper broke my printer.

cedorland, 2021-02-18:
> https://www.klipper3d.org/Contact.html#help-it-doesnt-work
> 
> Please try the Discord for help with initial setup.
Discord had no clue what went wrong or how to help. I'm unable to flash my printer manually, I'm going to try disassembling it now. It had never once had an issue before trying to install Klipper, so something in the original post points to how Klipper mangled my printer.

BlackStump, 2021-02-18:
Klipper doesn't mangle printers humans do that. seriously if you don't have the aptitude to following instructions then you should ask yourself why you didn't just leave your printer alone.  Don't blame Klipper for your own failings.

series8217, 2021-02-18:
It sounds like your USB/serial connection isn't working. This is likely coincidental to the Klipper install. @cedorland what is your username on the Klipper Discord?