Klipper3d/klipper #3968: Klipper goes into shutdown state after selecting extruder1 menu item

fleinze, 2021-02-20:
Hi, thank you for all the work you do on klipper!

I think I might have found a bug:
When I have a second extruder (extruder1) with shared heater defined in the config, a menu item apears on the display for ext1. When selecting this menu item the printer goes into shutdown immediately. 

Workaround: 
`[menu __main __temp __hotend1_target]
type: disabled`

Klipper Version: FIRMWARE_VERSION:v0.9.1-222-gdfd05251 FIRMWARE_NAME:Klipper

[klippy.log](https://github.com/KevinOConnor/klipper/files/6014475/klippy.log)

mcmatrix, 2021-02-20:
Hi @fleinze ,

The menu item `__hotend1_target` tries to evalute the extruder1 max_temp value from the config
but in case of shared heaters it's not defined.
`input_max: {printer.configfile.config.extruder1.max_temp}`

I'll check how to improve existing default menu to prevent this.

-Janar

mcmatrix, 2021-02-20:
Please check and test fix #3969 for this issue.

fleinze, 2021-02-20:
This fix works for my configuration. The extruder1 menu item does no longer show up.

closed 2021-02-20 by fleinze