Klipper3d/klipper #3973: Klipper overheats bed and never starts printing

gknops, 2021-02-20:
I can't get Klipper to print at lower bed temps: it overheats the bed by about 2.5ºC, but doesn't allow it to cool down (I keep seeing power being applied in Fluidd). I did run through a bed PID tuning, no help.

I tuned at 60ºC (my most used bed temp), but in this case I need 40ºC. It never drops below 42.5ºC. Ambient is around 20ºC.

[klippy-4.log](https://github.com/KevinOConnor/klipper/files/6016825/klippy-4.log)

gknops, 2021-02-21:
It is demonstrably an issue in Klipper and not with the configuration, and I followed all the rules. What a friendly place.

Gilabite, 2021-02-21:
clearly didnt follow anything, https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

ChipCE, 2021-02-25:
> It is demonstrably an issue in Klipper and not with the configuration, and I followed all the rules. What a friendly place.

The above msg is auto generated by github. Try go to discord first, you might get faster response there.

closed 2021-02-27 by KevinOConnor