Klipper3d/klipper #3976: Use PAUSE, RESUME, CANCEL_PRINT GCODE macros (if available) in LCD OctoPrint menu

TheJoshW, 2021-02-21:
Hi,
To ensure printer state is correctly saved/restored when using the LCD OctoPrint menu to pause/resume the print, it should use the PAUSE, RESUME, CANCEL_PRINT macros if available, similar to the sdcard menu implementation in menu.cfg.  Suggested changes to menu.cfg below:

```
[menu __main __octoprint __pause]
type: command
enable: {printer.idle_timeout.state == "Printing"}
name: Pause printing
gcode:
    {% if "pause_resume" in printer %}
        PAUSE
    {% else %}
        {action_respond_info('action:pause')}
    {% endif %}

[menu __main __octoprint __resume]
type: command
enable: {not printer.idle_timeout.state == "Printing"}
name: Resume printing
gcode:
    {% if "pause_resume" in printer %}
        RESUME
    {% else %}
        {action_respond_info('action:resume')}
    {% endif %}

[menu __main __octoprint __abort]
type: command
enable: {printer.idle_timeout.state == "Printing"}
name: Abort printing
gcode:
    {% if 'pause_resume' in printer %}
        CANCEL_PRINT
    {% else %}
        {action_respond_info('action:cancel')}
    {% endif %}

```

closed 2021-03-23 by github-actions[bot]

cfelicio, 2021-04-09:
I just wanted to say thank you, this code worked perfectly when added to printer.cfg to override the defaults. And now the menu works very well for me without Octoprint (I'm using Fluidd).

TheJoshW, 2021-04-09:
@cfelicio no worries. Glad it worked for you!