Klipper3d/klipper #3980: Internal error on command:"G1"

krpepe, 2021-02-22:
Good Morning everyone,

While printing this morning I tried running the following command and received internal error on command G1. 

 11:03:05
$ SYNC_STEPPER_TO_EXTRUDER STEPPER=spool_2 EXTRUDER=extruder
11:03:05
// Extruder stepper now syncing with 'extruder'
11:03:05
// Klipper state: Shutdown
11:03:05
!! Internal error on command:"G1" 

after looking in the log, this Is what I found 

stepcompress o=0 i=0 c=5 a=0: Invalid sequence
stepcompress o=0 i=0 c=5 a=0: Invalid sequence

I went to discord and they felt it was a bug so I came here to report it. klippy log is attached. Please let me know if you need anymore info.

Kevin
[klippy.log](https://github.com/KevinOConnor/klipper/files/6024012/klippy.log)

gardenali, 2021-02-22:
It looks like #3459 is related.

Disabling pressure advance seems to fix the issue for my printer.

I'm also having issues with this:
```
Stats 11438.9: gcodein=0 mcu: mcu_awake=0.003 mcu_task_avg=0.000021 mcu_task_stddev=0.000024 bytes_write=1317707 bytes_read=681175 bytes_retransmit=9 bytes_invalid=0 send_seq=30518 receive_seq=30518 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=883 freq=72001749 heater_bed: target=60 temp=60.1 pwm=0.165 sd_pos=93705 print_time=5010.452 buffer_time=2.214 print_stall=0 extruder: target=220 temp=220.1 pwm=0.497 sysload=0.09 cputime=336.750 memavail=757076
stepcompress o=2 i=0 c=10 a=0: Invalid sequence
stepcompress o=0 i=0 c=10 a=0: Invalid sequence
Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/gcode_move.py", line 141, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 181, in move
    self.toolhead.move(split_move, speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 420, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 186, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 336, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 295, in _update_move_time
    m.flush_moves(mcu_flush_time)
  File "/home/pi/klipper/klippy/mcu.py", line 779, in flush_moves
    % (self._name,))
error: Internal error in MCU 'mcu' stepcompress
Internal error on command:"G1"
virtual_sdcard dispatch
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 231, in work_handler
    self.gcode.run_script(lines[-1])
  File "/home/pi/klipper/klippy/gcode.py", line 200, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/gcode_move.py", line 141, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 181, in move
    self.toolhead.move(split_move, speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 420, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 186, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 336, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 295, in _update_move_time
    m.flush_moves(mcu_flush_time)
  File "/home/pi/klipper/klippy/mcu.py", line 779, in flush_moves
    % (self._name,))
error: Internal error in MCU 'mcu' stepcompress
Exiting SD card print (position 93705)
```

krpepe, 2021-02-22:
Just tested with pressure advance disabled and the gcode worked as intended. Defiantly related to PA.

@gardenali Thank you for the heads up!

gardenali, 2021-02-23:
That was a red herring. I had more of this with pressure_advance disabled:
`!! Internal error on command:"G1"`

```
grep -v "^Stats" klippy\ \(7\).log | tail ...

Got error 0 in read: (0)Success
Timeout with MCU 'mcu' (eventtime=6121.767894)
Transition to shutdown state: Lost communication with MCU 'mcu'
Dumping gcode input 0 blocks
gcode state: absolute_coord=True absolute_extrude=False base_position=[0.0, 0.0, -0.03, 1761.7737794000147] last_position=[36.907, 161.296, 0.44999999999999996, 3332.5465836000335] homing_position=[0.0, 0.0, -0.03, 0.0] speed_factor=0.0283333333333 extrude_factor=1.03 speed=68.0
Virtual sdcard (72704): 'E7.62182\nG1 X36.907 Y170.677 E0.02431\nG1 X158.677 Y48.907 E7.58743\nG1 X158.125 Y48.907 E0.02431\nG1 X36.907 Y170.125 E7.55305\nG1 X36.907 Y169.573 E0.02431\nG1 X157.573 Y48.907 E7.51867\nG1 X157.022 Y48.907 E0.02431\nG1 X36.907 Y169.021 E7.48428\nG1 X36.907 Y168.470 E0.02431\nG1 X156.470 Y48.907 E7.44990\nG1 X155.918 Y48.907 E0.02431\nG1 X36.907 Y167.918 E7.41552\nG1 X36.907 Y167.366 E0.02431\nG1 X155.366 Y48.907 E7.38113\nG1 X154.814 Y48.907 E0.02431\nG1 X36.907 Y166.814 E7.34675\nG1 X36.907 Y166.262 E0.02431\nG1 X154.262 Y48.907 E7.31237\nG1 X153.711 Y48.907 E0.02431\nG1 X36.907 Y165.711 E7.27798\nG1 X36.907 Y165.159 E0.02431\nG1 X153.159 Y48.907 E7.24360\nG1 X152.607 Y48.907 E0.02431\nG1 X36.907 Y164.607 E7.20922\nG1 X36.907 Y164.055 E0.02431\nG1 X152.055 Y48.907 E7.17483\nG1 X151.503 Y48.907 E0.02431\nG1 X36.907 Y163.503 E7.14045\nG1 X36.907 Y162.952 E0.02431\nG1 X150.952 Y48.907 E7.10607\nG1 X150.400 Y48.907 E0.02431\nG1 X36.907 Y162.400 E7.07168\nG1 X36.907 Y161.848 E0.02431\nG1 X149.848 Y48.907 E7.03730\nG1 X149.296 Y48.907 E0.02431\n'
Upcoming (73728): 'G1 X36.907 Y161.296 E7.00292\nG1 X36.907 Y160.744 E0.02431\nG1 X148.744 Y48.907 E6.96853\nG1 X148.192 Y48.907 E0.02431\nG1 X36.907 Y'
Reactor garbage collection: (6098.035954705, 5901.923455352, 0.0)
stepcompress o=0 i=0 c=14 a=0: Invalid sequence
Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/gcode_move.py", line 141, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 181, in move
    self.toolhead.move(split_move, speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 420, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 186, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 336, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 295, in _update_move_time
    m.flush_moves(mcu_flush_time)
  File "/home/pi/klipper/klippy/mcu.py", line 779, in flush_moves
    % (self._name,))
error: Internal error in MCU 'mcu' stepcompress
Internal error on command:"G1"
virtual_sdcard dispatch
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 231, in work_handler
    self.gcode.run_script(lines[-1])
  File "/home/pi/klipper/klippy/gcode.py", line 200, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/gcode_move.py", line 141, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 181, in move
    self.toolhead.move(split_move, speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 420, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 186, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 336, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 295, in _update_move_time
    m.flush_moves(mcu_flush_time)
  File "/home/pi/klipper/klippy/mcu.py", line 779, in flush_moves
    % (self._name,))
error: Internal error in MCU 'mcu' stepcompress
Exiting SD card print (position 73728)
stepcompress o=0 i=0 c=14 a=0: Invalid sequence
Exception in flush_handler
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/toolhead.py", line 395, in _flush_handler
    self.flush_step_generation()
  File "/home/pi/klipper/klippy/toolhead.py", line 340, in flush_step_generation
    self.move_queue.flush()
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 336, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 295, in _update_move_time
    m.flush_moves(mcu_flush_time)
  File "/home/pi/klipper/klippy/mcu.py", line 779, in flush_moves
    % (self._name,))
error: Internal error in MCU 'mcu' stepcompress

krpepe, 2021-02-23:
@gardenali I have never had a problem until I tried issuing the SYNC_STEPPER gcode command. Everything just worked. This is a first for me and I have been running Klipper for nearly 2 yrs.

krpepe, 2021-02-26:
[klippy.log](https://github.com/KevinOConnor/klipper/files/6052461/klippy.log)
 @KevinOConnor  I could not hit e stop after this error. It dead stops and says needs firmware restart. should be last entry in the log. I will post a picture of my screen immediately after the error.
[image]

After I rebooted the host and did a firmware restart I hit the estop. Not sure if this will help but below 
[klippy.log](https://github.com/KevinOConnor/klipper/files/6052481/klippy.log)
is that klippy.log.

KevinOConnor, 2021-02-27:
Thanks.  I think it should be fixed now (commit c5a9d791).

-Kevin

krpepe, 2021-02-27:
@KevinOConnor Thank you!! I will update and let you know! Have a great weekend and thank you for all of your hard work!
Kevin

krpepe, 2021-02-27:
@KevinOConnor Thank you again!! Works perfectly! I will close the issue.

Have a great weekend,
Kevin

closed 2021-02-27 by krpepe