Klipper3d/klipper #3981: Klipper process doesn't survive FIRMWARE_RESTART with configured raspberry MCU

ScotchTape, 2021-02-23:
When editing printer,cfg in one of the WebUIs and issue a Save&Restart or FIRMWARE_RESTART klippy exits and needs to be restarted manually when a Raspberry Pi is configured as additional MCU. Once it's removed a FIRMWARE_RESTART comes back as expected.
I'm using a Duet2 Wifi and, to be sure, setup a new image yesterday and using the most recent pull from today.
The klippy.log and the journal is attached with both cases.

[journal.zip](https://github.com/KevinOConnor/klipper/files/6026246/journal.zip)
[klippy.log](https://github.com/KevinOConnor/klipper/files/6026247/klippy.log)

dalegaard, 2021-02-26:
I believe this happens because Klippers gets killed by a `SIGHUP`, presumably because `klipper_mcu` took its pseudo-terminal with it when it restarted. I've been launching `klippy` with `nohup` to avoid this issue, as I wasn't sure if this was a problem on my setup or a general problem and didn't have time to dig into this deeper. This "fixed" the problem by forcing Klippy to restart when it encounters an error when trying to talk to `klipper_mcu`.

ScotchTape, 2021-02-26:
Ok, thanks for the tip, I’ll give it a try. 
But regardless if the workaround fixes it, it should be fixed, since I’m not the only one with this problem, according to the official discord. :)

dalegaard, 2021-02-27:
Since this is affecting me, I'll take another crack at it!

Sineos, 2021-02-27:
Can confirm this behavior. I created a `service klipper restart` link in octoprint to easily work around this

dalegaard, 2021-03-01:
This should be fixed with the merge of #3996

ScotchTape, 2021-03-01:
I can confirm it's fixed for me, reverted the nohup fix in the unit file, rebuild the linux process, installed it and a FIRMWARE_RESTART works as expected.
I leave it open for now.

Thanks

closed 2021-03-03 by ScotchTape