Klipper3d/klipper #3986: Update to v0.9.1-261 - Printer not ready

Matrixer007, 2021-02-24:
Hello!
I made an update on my Ender 3 with MKS Gen 1.4 and not the pinter would not work. I made the changes in the printer.cfg after looking in Config_Changes.md. Ist does'nt work, there is the message: Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.
Klipper state: Not ready
With 0.8.0 the printer works fine.
Can you help me?
Thanks

klippy.log is attached....

theopensourcerer, 2021-02-24:
No log attached, but also, GitHub is not really the correct place for these kind of questions... 

Please see this page for more information https://www.klipper3d.org/Contact.html

Matrixer007, 2021-02-24:
Okay, thanks.

closed 2021-02-24 by Matrixer007