Klipper3d/klipper #3988: Stepper Motors dont move after installing Klipper on Duet

StefanX13, 2021-02-24:
Setup:
Full Working BLV MGN 
Board: FYSETC Duet2 Wifi
Motors: StepperOnline 1,8 deg 2Amps
Klipper Version: "last"
MainSail and Moonraker
Installed with KIAUH on fresh RasbianLite on RPi 3B
2 PSUs switched by SSA with PSUon on Pin PD15
Working: The endstops show real status, Temperature readings are fine, Fans are working after switch PSU on, Noozle Heat and Bed Heat seams to work but a little bit slow compared to behaviour on RRF(maybe just my feelings).

Problem: After flashing clipper and making a new config based on the generic-duet2.cfg the Stepper Motors, which are working fine with RRF until flashing Klipper bearly moves at all. When try with STEPPER_BUZZ STEPPER=stepper_x (as well as with Y or Z/Z1) the Motor jerk and i can hear a whistle but no real movement at all. When i try to home its the same behaviour until the homing fails with timeout.

Additional Infos: I already tried a complete new Klipper installation on a fresh Raspbian Lite, flashed Klipper 3 times new.

Dont know how to prefer. I realy would like to use Klipper on my BLV MGN since i use it on my i3Mega and really like the config with ADXL345, but i dont see any chance at the moment.

Dumb Question: Is it possible that there is an issue with PSU_On and the SSR is switching on/of in Milliseconds? Maybe dumb question, but this would explain the behavoir since the steppers are working like they havent enough amps. This would also explain the slower(?)  heating 

[klippy (8).log](https://github.com/KevinOConnor/klipper/files/6036975/klippy.8.log)

paulhew69, 2021-02-25:
Under RRF were your M906  for the steppers 1.6a? If so way too high if using klipper.
You need to do the calc to bring them inline.
The calculation of rms from peak current is this:

Motor_Peak_Current x 0.707

And since we dont wanna push the motors with 100% we do:

Motor_Peak_Current x 0.707 x 0.80

that will make a 20% safety margin.

so a stock ender 3 motor that is 0.84A peak 

0.84x0.707x0.8 = 0.47A 

so you put 0.47 in run_current of your tmc stepper. 

A normal 1.4A stepper is:

1.4x0.707x0.8 = 0.79A

so you put 0.79 in run_current of your tmc stepper.

StefanX13, 2021-02-25:
Since my klipper related problem cant be solved, since it seems to be invalid according to the developer i close this.
PS: The printer runs now.
PPS: Its a klipper related problem/Bug with the duet, at least with the FYSETC.
Closed

closed 2021-02-25 by StefanX13