Klipper3d/klipper #3989: Move exceeds maximum extrusion nonsense - Ender with BMG extruder & SKR E3 2.0

tarsonis123, 2021-02-24:
Could not once print with Klipper as I always get "Move exceeds maximum extrusion (0.647mm^2 vs 0.640mm^2)" after the prime start code. Slicer is Cura 4.8. Printer is an Ender 3 with MicroSwiss and BMG. The example config for the SKR E3 board doesn't have _max_extrude_cross_section and _max_extrude_cross_section but I did try those.

I tried **_max_extrude_cross_section: 50_** and the BMG went berserk with me hitting Emergency stop...

The fun part is which lets me think my Klipper is dumb af. If I change the value **_max_extrude_cross_section: 0.7_**
Then I get:
**_Move exceeds maximum extrusion (0.718mm^2 vs 0.700mm^2)_**

I set the value to 0.8 and I get:

**_Move exceeds maximum extrusion (0.859mm^2 vs 0.800mm^2)_**

The docs says the gear_ratio is 3:1 for the BMG but then I find comments to use 50:17 for the BMG. And then the step_distance should be calculated differently (not mentioning how..). But step_distance is now rotation_distance. The docs and calculations are cryptic to me as non-DEV head.

My calibrated E-steps with Marlin were 404.40 with bowden. If I did the calculation right the value should be rotation_distance: 7.9129 then?

When I do an extrude test with 100mm marked on the filament the extruded material is 100mm.

My latest extruder config: I flipped around every possible value I found on the net fitting my printer specs without any success
```
[extruder]
step_pin: PB3
dir_pin: !PB4
enable_pin: !PD2
microsteps: 16
rotation_distance: 7.9129
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA0
control: pid
pid_Kp: 23.760
pid_Ki: 1.20
pid_Kd: 117.612
min_temp: 0
max_temp: 260
max_extrude_cross_section: 0.8
max_extrude_only_distance: 500
```

My full config is:

https://www.dropbox.com/s/1tg9vleayfmfdne/printer.cfg?dl=0

File I'm trying to print (I did test other files):

https://www.dropbox.com/s/nxp2uwrs9p5apvs/First_layer_test_box_25x25x1%2C5_0_PLA_0.2mm_94%25_0h_1m.gcode?dl=0
[klippy.log](https://github.com/KevinOConnor/klipper/files/6037640/klippy.log)

th33xitus, 2021-02-24:
Configuration issues are no valid reason to open a github issue.

You disregarded the following notice when you wrote your post 
``
Therefore this issue is likely to be closed soonish.

Please have a look at the linked website and read carefully: https://www.klipper3d.org/Contact.html

tarsonis123, 2021-02-24:
I did. This is a bug and not a configuration issue jeez...

jakep82, 2021-02-24:
It sounds like a configuration issue with absolute versus relative positioning for the extruder.

mental405, 2021-02-24:
It is your start print macro.

tarsonis123, 2021-02-24:
> Configuration issues are no valid reason to open a github issue.

If I set a value X accordingly to the firmware and it replies with X reached a maximum and I changed the maximum higher than the reported value X and I get another reply the new value reached a maximum, and so on. I consider this a bug running circles. At least usability wise.  

> It is your start print macro.
I did try my own Marlin start code which of course worked fine and one from here with the same behaviour. The start code you posted isn't what I'm using.  This is the prime code I'm using.  

 ; Now print a line of filament to prepare extrusion
    G91
    G1 F150 E5 ; Extrude 15mm in the air
    G90
    G1 Y10 F3000  ;Move Y to knock the filament off
    M400
    G1 Z0.2 F500 ; Drop the Z Back
    G91
    G1 Y100 F750 E3 ; Extrude to the build plate
    G90
    G1 Y110 F4000 ; Quick Wipe
    G1 Z2.5 F1500 ; Lift the Z
    G92 E0
```

jakep82, 2021-02-24:
The issue is in fact your start_print macro.  The last line is M83 which sets the extruder to relative mode.  Your slicer is generating gcode assuming absolute mode.  Delete the M83 from your macro, and I think you'll find that Klipper is working just fine.

mental405, 2021-02-24:
on the last line of the macro it changes to relative in the gcode file after the start print macro 
it resets the extruder distance
it retracts 2mm
it then starts to print and fails.

again, not a bug.

tarsonis123, 2021-02-24:
> The issue is in fact your start_print macro. The last line is M83 which sets the extruder to relative mode. Your slicer is generating gcode assuming absolute mode. Delete the M83 from your macro, and I think you'll find that Klipper is working just fine.

Ok, thanks a lot! It was that. I must have gotten an old config from the guy as he also had step_distance as value in it and claimed it works perfectly for him.

> So again, this is not a bug.
Well call it what you want. I'm graphic/ux designer and worked with devs on daily base for more than decade. On the conscious level devs see every issue on the user side but 90% of the cases the resulting product should not be used soley by devs. If the usability return for the user is fcked he doesn't pay the bills for pure rational thinking mind of the dev.

jakep82, 2021-02-24:
Not sure what you're looking for here.  You attacked the devs in the original post by saying Klipper was "dumb af", only to find out you blindly copy/pasted a non-working config from someone else.  Just take the L and move on.

th33xitus, 2021-02-24:
@tarsonis123 Im sure you can revise that statement by now...
> This is a bug and not a configuration issue jeez... Would you have read my thread would you have seen that I configured it accordingly to the docs

I think we know all know its truly not a bug... Typical case of PEBCAK.
Please close this issue now to mark it as resolved.

theopensourcerer, 2021-02-24:
You learn something new everyday! Never come across of PEBCAK before :-)

jakep82, 2021-02-24:
> I tried **_max_extrude_cross_section: 50_** and the BMG went berserk with me hitting Emergency stop...

I don't want to beat a dead horse, but I want to say one more thing for anyone else that sees this thread in the future.  Please don't disable safety features in Klipper when an error is occurring for reasons you don't understand.  In this case there were no real consequences, but a 3D printer can be a dangerous thing.  If an error is occurring, don't disable the safety checks to get around it.  Find the root cause and fix it.