Klipper3d/klipper #3990: Cannot connect to duet2 wifi

BastianGschrey, 2021-02-24:
Hello,

I cannot connect to duet2 Wifi.
I flashed the firmware to the MCU succesfully using 	SAM3/SAM4 (Due and Duet) 
		Processor model (SAM4e8e (Duet Wifi/Eth))  --->  SAM4e8e (Duet Wifi/Eth)

Flashing return successfully:
Entering bootloader on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0
Device reconnect on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0
lib/bossac/bin/bossac -U -p /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:                                                                1.0 -e -w out/klipper.bin -v

Erase flash

Done in 4.377 seconds
Write 19780 bytes to flash (39 pages)
[==============================] 100% (39/39 pages)
Done in 0.298 seconds
Verify 19780 bytes of flash
[==============================] 100% (39/39 pages)
Verify successful
Done in 0.294 seconds
Device reconnect on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0

ls /dev/serial/by-id returns usb-03eb_6124-if00
so I used /dev/serial/by-id/usb-03eb_6124-if00 in my printer.cfg, I have also tried /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0

klippy.log says:
```
========= Last MCU build config =========
# CONFIG_LOW_LEVEL_OPTIONS is not set
# CONFIG_MACH_AVR is not set
CONFIG_MACH_ATSAM=y
# CONFIG_MACH_ATSAMD is not set
# CONFIG_MACH_LPC176X is not set
# CONFIG_MACH_STM32 is not set
# CONFIG_MACH_PRU is not set
# CONFIG_MACH_LINUX is not set
# CONFIG_MACH_SIMU is not set
CONFIG_STEP_DELAY=2
CONFIG_BOARD_DIRECTORY="atsam"
CONFIG_MCU="sam4e8e"
CONFIG_CLOCK_FREQ=120000000
# CONFIG_USBSERIAL is not set
CONFIG_SERIAL=y
CONFIG_ATSAM_SELECT=y
# CONFIG_MACH_SAM3X8E is not set
# CONFIG_MACH_SAM3X8C is not set
# CONFIG_MACH_SAM4S8C is not set
CONFIG_MACH_SAM4E8E=y
CONFIG_MACH_SAM4=y
CONFIG_MACH_SAM4E=y
CONFIG_FLASH_START=0x400000
CONFIG_FLASH_SIZE=0x80000
CONFIG_RAM_START=0x20000000
CONFIG_RAM_SIZE=0x20000
CONFIG_STACK_SIZE=512
CONFIG_SERIAL_BAUD=250000
CONFIG_USB_VENDOR_ID=0x1d50
CONFIG_USB_DEVICE_ID=0x614e
CONFIG_USB_SERIAL_NUMBER_CHIPID=y
CONFIG_USB_SERIAL_NUMBER="12345"
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
CONFIG_HAVE_GPIO_I2C=y
CONFIG_HAVE_GPIO_HARD_PWM=y
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_HAVE_STRICT_TIMING=y
CONFIG_HAVE_CHIPID=y
CONFIG_INLINE_STEPPER_HACK=y
=======================
Build file /home/pi/klipper/klippy/../out/klipper.dict(6486): Wed Feb 24 22:14:03 2021
Last MCU build version: v0.9.1-261-ga8742e98-20210224_221403-klipper-ht
Last MCU build tools: gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1
Last MCU build config: BUS_PINS_twi1=PB5,PB4 BUS_PINS_twi0=PA4,PA3 BUS_PINS_spi=PA12,PA13,PA14 SERIAL_BAUD=250000 BUS_PINS_usart0=PB0,PB1,PB13 STEP_DELAY=2 ADC_MAX=4095 BUS_PINS_usart1=PA21,PA22,PA23 PWM_MAX=255 CLOCK_FREQ=120000000 RECEIVE_WINDOW=192 MCU=sam4e8e STATS_SUMSQ_BASE=256 RESERVE_PINS_serial=PA9,PA10
Build file /home/pi/klipper/klippy/../out/klipper.elf(162360): Wed Feb 24 22:14:09 2021
Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 188, in send_with_response
    return src.get_response(cmd, self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 242, in get_response
    self.serial.raw_send_wait_ack(cmd, minclock, reqclock, cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 180, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed

```

paulhew69, 2021-02-25:
My Duet2 Eth reports
[mcu]
serial: /dev/serial/by-id/usb-Klipper_sam4e8e_00313753384C39323132303239303534-if00

This works, well.

Power off, count 10, power on, jumper the Erase on the board for 4 secs, re-flash Klipper.

I have notes in the Klipper discord, let me find them for you. Not teaching you to suck eggs, but it took me a while.

paulhew69, 2021-02-25:
Found my notes
Notes for updating firmware on Duet2Eth board from a Pi. Could this be pinned?
1) Shutdown Klipper using - sudo service klipper stop
2) Jumper the pins on the Duet board to erase the flash, then remove jumper
3) run 'make makemenuconfig' Make sure board is selected correctly and then exit.
4) run 'make' to build the firmware
5) run 'ls /dev/serial/by-id/' - it will look like this - '/dev/serial/by-id/usb-03eb_6124-if00 as there is no firmware running.
6) Then run, 'make flash FLASH_DEVICE=/dev/serial/by-id/usb-03eb_6124-if00' - replacing the part after by-id with your result from action 5.
It will give you something similar to this.
  Flashing out/klipper.bin to /dev/serial/by-id/usb-03eb_6124-if00
Entering bootloader on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0
Device reconnect on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0
lib/bossac/bin/bossac -U -p /dev/serial/by-path/platform-3f980000.usb-usb-0:1.3:1.0 -e -w out/klipper.bin -v

Erase flash

Done in 4.353 seconds
Write 21100 bytes to flash (42 pages)
[==============================] 100% (42/42 pages)
Done in 0.321 seconds
Verify 21100 bytes of flash
[==============================] 100% (42/42 pages)
Verify successful
Done in 0.315 seconds

7) Restart Klipper using - 'sudo service klipper start`

Job Done!

closed 2021-02-28 by github-actions[bot]