Klipper3d/klipper #3991: Ability to probe 4 points of the bed to assist with X Rail Alignment (RailCore ZLT)

paulhew69, 2021-02-25:
RailCore are primarily built with Duet boards and RRF
We have a Bed Mesh routine that assists us in aligning the X Rails. It homes, trams or levels the bed, then creates a heightmap with Using 4 points, which allows us to tweak the rails as needed.
When then run until we are happy with the end result.

I tried to do this with Klipper but Klipper complains about the lack of probe points.
Ideally to be able to tell Klipper where we want to probe for RailAlignment would be helpful and to create a mesh which we can view.

I have to keep flipping between RRF and Klipper if I think there is an issue with the rails.
Not often but enough to warrant creating an Issue.

Log attached just to keep Gitissuebot quiet!
[klippy(2).log](https://github.com/KevinOConnor/klipper/files/6042391/klippy.2.log)

jakep82, 2021-02-25:
https://www.klipper3d.org/Contact.html#i-have-a-feature-request

paulhew69, 2021-02-25:
Surely this is not a feature request? Invalid?
It is something that is already inbuilt but does not work correctly. Bed mesh is not a new feature, so how can it be a feature request?

jakep82, 2021-02-25:
As explained in the docs, bed_mesh requires a minimum of 9 points.  I'm not sure what you're trying to accomplish as I'm not familiar with this printer, but bed_tilt or z_tilt might be better options.  If those aren't good options, then it would seem you're requesting a feature that doesn't currently exist.

paulhew69, 2021-02-25:
I appreciate what you are saying, I use all of those functions, but if for some reason it fails to tram, it normally indicates that the rails could of shifted either up or down.
This is why I had to reload RRF firmware to check my rails and they _were_ out.
It is why there are XRAT prints on thingiverse. X Rail Alignment Tool, which allow people to adjust finely the rails in relation to the other.

In RRF we are able to have a macro that effectively does a  bed mesh but with only 4 points at our maximum safe values and displays it as a heightmap as four points, which allows use to visualize the tweaks we need to make to the rails and also which end of the rail(s).
M557 X45:275 Y46:275 P2:2 this defines our minimum and max probe area and how many points (P) to probe.

RailCores run 2 'X' rails so without this ability in RRF I do not know how I would be able to align the rails to each other.
It does not matter how well the machine is built, square etc. If the extrusions are out by a fraction how would you know if you cannot see it?

So it is not a feature request, just bed_mesh has a limitation in its minimum allowed points.
.
I run Klipper on 2 other machines but would like to be able to run it on all 3. The other 2 are Vorons.
I like the Klipper, Moonraker and MainSail integration and like how it works with the Duet boards with minimum fuss.
Hence the question.
Regards,
Paul.

jakep82, 2021-02-25:
The bot will close this in a couple days, so you'll have better luck following the recommendation and inquiring on the Klipper Discord.

paulhew69, 2021-02-25:
I am trying to be informative and constructive.

I Found an issue and that is your best reply?
Sounds like, 'jog on, can't be bothered'

Regards,
Paul Hewson
Network Specialist

jakep82, 2021-02-25:
>I Found an issue and that is your best reply? 

Klipper is working as intended.  The issue you're describing is that Klipper lacks a feature you want.  If you want to know why Github is no longer used for support requests, please read [this PR.](https://github.com/KevinOConnor/klipper/pull/3762)

closed 2021-03-02 by KevinOConnor