Klipper3d/klipper #3993: "MCU 'mcu' shutdown: Timer too close"

thebrettman73, 2021-02-25:
I have an Ender 5 with the 1.1.4 board that I'm getting the "MCU 'mcu' shutdown: Timer too close" error intermittently. I have tried:

-3 different USB cables but it still happens. The current one is 300 mm so I don't think length of cable is my issue at this point.

-I had a secondary MCU set up to run a filament sensor directly connected to the Raspberry Pi. I disabled this for now, but I've still gotten the error.

-The VCC/Power pin on the USB has been taped.

-I had run OctoPrint with Marlin for months without any communication issues on the original USB cord.

Are there other troubleshooting steps that I can try to track this down? I'm thinking I'm missing something simple and it's probably in the documentation somewhere, but I'm coming up short...

[klippy (7).log](https://github.com/KevinOConnor/klipper/files/6043251/klippy.7.log)

thebrettman73, 2021-02-25:
It happened again, so I have added the updated log file:

[klippy (8).log](https://github.com/KevinOConnor/klipper/files/6044009/klippy.8.log)

KevinOConnor, 2021-02-27:
Both errors are occurring because the host thread that handles temperature updates failed to get sufficient CPU time.  This usually occurs because of a system event of some kind (high swap usage, host disk errors, excessive cpu usage by other processes, host voltage fluctuations, etc.).  The logs indicate some high host cpu load and low system memory - not enough to directly cause an issue, but indicative of something else using notable system resources.  Unfortunately, the logs don't give insight into the root cause.

If you're familiar with Linux you can look in the system logs at the time of the event and see if they reveal anything useful.

-Kevin

sdavis175, 2021-03-02:
I am also having the same issue, my logs are attached. Not familiar with Linux logs, any recommendations on what to look at?
Here is my load graph I generated after the issue occurred multiple times:
[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/6069365/klippy.log)
[moonraker.log](https://github.com/KevinOConnor/klipper/files/6069366/moonraker.log)

thebrettman73, 2021-03-02:
I haven't had it happen again since the second log file upload. I'm still trying to track down what may be causing the issue.

darkpoetry, 2021-03-11:
I run three Klipper printers on a very powerful server (in a kvm/qemu container) with Octoprint and I've been getting these randomly too of late. On investigation I noticed that Octoprint has taken to devouring huge amounts of memory, to the point that Klipper may be getting starved when the system starts swapping. If you're using Octoprint, see how much memory it's eating with top. Might need to be regularly manually restarted. I suspect that it's failing to free up memory used after the upload of large prints or something.

thebrettman73, 2021-03-12:
I haven't had a reoccurrence since the last log file was uploaded. The only change I have made was to disable "power save".  I'm unsure if/why that was the problem, but everything has been working as expected since. Directions to disable found here: [https://github.com/fordsfords/wlan_pwr](https://github.com/fordsfords/wlan_pwr)

sdavis175, 2021-03-13:
I am running fluidd, not octoprint for my single klipper instance. For disabling power save, I am following the github instructions you linked however I am running into an error with not being able to access a file from this line:
`sudo wget -O/etc/network/if-up.d/wlan_pwr http://fordsfords.github.io/wlan_pwr/wlan_pwr`
The error:
`--2021-03-13 07:14:10--  http://fordsfords.github.io/wlan_pwr/wlan_pwr
Resolving fordsfords.github.io (fordsfords.github.io)... 185.199.108.153, 185.199.111.153, 185.199.109.153, ...
Connecting to fordsfords.github.io (fordsfords.github.io)|185.199.108.153|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://www.geeky-boy.com/wlan_pwr/wlan_pwr [following]
--2021-03-13 07:14:10--  http://www.geeky-boy.com/wlan_pwr/wlan_pwr
Resolving www.geeky-boy.com (www.geeky-boy.com)... 185.199.108.153, 185.199.109.153, 185.199.110.153, ...
Reusing existing connection to fordsfords.github.io:80.
HTTP request sent, awaiting response... 404 Not Found
2021-03-13 07:14:10 ERROR 404: Not Found.`
Looks like the github page I am trying to download from is down. Any suggestions?

thebrettman73, 2021-03-13:
I didn't run into any issues when I did it, so I'm not sure what may be the issue. Also, I can't say for sure that was the issue, only that it hasn't happen again, so YMMV...

Sineos, 2021-03-13:
> I am running fluidd, not octoprint for my single klipper instance. For disabling power save, I am following the github instructions you linked however I am running into an error with not being able [...]

Well you could just copy the content of the file or use this link: https://gitcdn.link/repo/fordsfords/wlan_pwr/gh-pages/wlan_pwr

thebrettman73, 2021-03-19:
I got the "Timer too close" error again. I have looked at the syslog and can see the error, but I don't see the problem (or don't understand what I'm looking at). If anyone has some insight, I would appreciate it. Thanks!
[syslog.txt](https://github.com/KevinOConnor/klipper/files/6173453/syslog.txt)
[klippy (13).log](https://github.com/KevinOConnor/klipper/files/6173454/klippy.13.log)

closed 2021-05-02 by github-actions[bot]