Klipper3d/klipper #3999: New TMC Code: Error on homing

vertigo235, 2021-02-27:
My setup doesn't like this merge, get this error. 

`TMC 'stepper_z' reports error: DRV_STATUS: ffffffff SG_RESULT=1023 fsactive=1 CS_ACTUAL=31 stallGuard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown`

EISNY RETRO TMC2130

_Originally posted by @vertigo235 in https://github.com/KevinOConnor/klipper/issues/3974#issuecomment-787158108_

ShohninDmitriy, 2021-02-28:
The same problem 
[klippy.log](https://github.com/KevinOConnor/klipper/files/6055955/klippy.log)

vertigo235, 2021-02-28:
[klippy.log](https://github.com/KevinOConnor/klipper/files/6055995/klippy.log)

KevinOConnor, 2021-02-28:
The micro-controller is not able to read from the tmc driver.  This could be for a variety of reasons - something else may be interfering with the control SPI messages, the micro-controller may not be communicating with the tmc driver all and it is actually running in standalone mode, or there is a wiring problem to that particular tmc driver.

If you run DUMP_TMC on each of your drivers, you should not see an `ffffffff` response - that is a symptom of not being able to communicate with the driver.

Since you have an einsy rambo, try adding the following to the config:
```
[static_digital_output sdcard_cs_disable]
pins: PC5, PJ6
```

-Kevin

KevinOConnor, 2021-02-28:
Actually since you're on the einsy retro, you should try this in the config:
```
[static_digital_output sdcard_cs_disable]
pins: PB0
```

vertigo235, 2021-02-28:
I do not understand what this did, but it does seem to have fixed the issue.

closed 2021-02-28 by vertigo235