Klipper3d/klipper #4006: Ender 3 SKR mini E3 unable to read tmc uart 'stepper_z' register "gstat"

Actispeedy, 2021-03-01:
Hello together,

ive bought the SKR mini E3 v2 and after installing klipper on it, everything works fine (sensorless homing, bltouch ) but if i try to print, there comes randomly the "unable to read tmc uart 'stepper_z' register "drv_status"" sometimes this 
"unable to read tmc uart 'stepper_z' register "gstat"" and sometime  it is a other stepper then z
the issue comes random after 3 -4 layers or sometimes after the first print line.
i dont know whats wrong... and sorry for the bad english
[klippy.log](https://github.com/KevinOConnor/klipper/files/6059106/klippy.log)

KevinOConnor, 2021-03-01:
There were some changes in this area of the code recently.  You should pull the latest version of the code and retry (see the FAQ for upgrade instructions).

-Kevin

Actispeedy, 2021-03-01:
maybe im too old for this, but i cant find any FAQ for upgrade instructions. (sorry for this) 
but i found this: https://github.com/KevinOConnor/klipper/pull/1957/commits/f76d1f5e4c7b9f1106cf375694184b43932cdd0f#diff-92f8e1ec2c8a8f0a80a4185a061eb0c4b5e6778bc42bc8c6ead03006fdfec716
and it also doesnt work. There comes the "Unknown pin chip name 'tmc2209_stepper_x'" message

Actispeedy, 2021-03-01:
if i copy&paste the wohle code, then there comes this message : 
Internal error during connect: __init__() takes at least 4 arguments (3 given)

KevinOConnor, 2021-03-01:
Upgrade instructions are at https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software

The github issue tracker isn't a good place to request help - see: https://www.klipper3d.org/Contact.html

-Kevin

Actispeedy, 2021-03-03:
sorry for that. 
but im realy sure that i have installed the latest versions. it was clean new octopi and klipper installation

closed 2021-03-04 by Actispeedy