Klipper3d/klipper #4009: G4 issue?

robustini, 2021-03-02:
I am trying to time the plate cooling using the G4 command.
Unfortunately, it seems to me that it doesn't work.
Here is an example, considering a bed temperature of 110° at the end of the print: 

M140 S90 ; bed temp at 90°
G4 S180 ;wait 180 milliseconds
M140 S80
G4 S180
M140 S70
G4 S180
M140 S60
G4 S180
M140 S50
G4 S180
M140 S0 ; turn off bed

"S" should be a value in seconds.
Instead of running the g-code correctly Klipper when it executes this sequence immediately turns off the bed.
Am I wrong or is Klipper wrong?

jakep82, 2021-03-02:
The docs only list support for the P variable. 

https://www.klipper3d.org/G-Codes.html

robustini, 2021-03-03:
> 
> 
> The docs only list support for the P variable.
> 
> https://www.klipper3d.org/G-Codes.html

This is a problem, I'll try to solve it in another way.

closed 2021-03-03 by robustini