Klipper3d/klipper #4013: SET_SERVO occasionally moves multiple servos

Block137, 2021-03-03:
[klippy+M112.zip](https://github.com/KevinOConnor/klipper/files/6074109/klippy%2BM112.zip)

I have 2 servos configured (KST DS215MG and KST DS589MG). They appears to work properly but occasionally, after a SET_SERVO command, either they don't respond or both moves(at the same time despite having SERVO= assigned) to incorrect positions. Sending another SET_SERVO command to any servo corrects both servos back to their commanded positions.

I'm unable to reproduce this behavior reliably yet, it appears every 10~20 SET_SERVO commands.

The 2 servos were initially assigned to PC4 and PC5 of SKR-PRO-V1.2 (STM32F407). Moving them to different ports (PC5 and PF9) and different boards (PC5 and board2:PC5) did not solve it.

KevinOConnor, 2021-03-04:
It sounds like a hardware issue of some kind.  This is likely not the best place to report it - see: https://www.klipper3d.org/Contact.html

-Kevin

Block137, 2021-03-05:
Unable to reproduce it anymore, likely a glitch, closing.

closed 2021-03-05 by Block137