Klipper3d/klipper #4014: No connection anymore

RcTomcat1, 2021-03-03:
Hi Guys,

after about two weeks of using klipper on an eryone thinker s i can't get a connection anymore.
Setup:
Pi 4
Eryone Thinker S

I get the following error message in the MainsailOS GUI:

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.

Klipper Log Shows:
Unable to open serial port: [Errno 2] could not open port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0'

Things i did:
Checked serial via  sudo ls /dev/serial/by-id/*
Result: ls: cannot access '/dev/serial/by-id/*': No such file or directory

Changed the USB Cable three times with cables that work on other printers (octoprint)
Same result, no serial.

ls/dev shows 

autofs         hwrng         mem        ram14           serial1  tty18  tty33  tty49  tty7       vcs5      vga_arbiter
block          initctl       mmcblk0    ram15           shm      tty19  tty34  tty5   tty8       vcs6      vhci
btrfs-control  input         mmcblk0p1  ram2            snd      tty2   tty35  tty50  tty9       vcsa      video0
bus            kmsg          mmcblk0p2  ram3            stderr   tty20  tty36  tty51  ttyAMA0    vcsa1     video10
cachefiles     log           mqueue     ram4            stdin    tty21  tty37  tty52  ttyprintk  vcsa2     video11
char           loop0         net        ram5            stdout   tty22  tty38  tty53  uhid       vcsa3     video12
console        loop1         null       ram6            tty      tty23  tty39  tty54  uinput     vcsa4     video13
cuse           loop2         port       ram7            tty0     tty24  tty4   tty55  urandom    vcsa5     video14
disk           loop3         ppp        ram8            tty1     tty25  tty40  tty56  v4l        vcsa6     video15
dma_heap       loop4         ptmx       ram9            tty10    tty26  tty41  tty57  vchiq      vcsm-cma  video16
dri            loop5         pts        random          tty11    tty27  tty42  tty58  vcio       vcsu      watchdog
fd             loop6         ram0       raw             tty12    tty28  tty43  tty59  vc-mem     vcsu1     watchdog0
full           loop7         ram1       rfkill          tty13    tty29  tty44  tty6   vcs        vcsu2     zero
fuse           loop-control  ram10      rpivid-h264mem  tty14    tty3   tty45  tty60  vcs1       vcsu3
gpiochip0      mapper        ram11      rpivid-hevcmem  tty15    tty30  tty46  tty61  vcs2       vcsu4
gpiochip1      media0        ram12      rpivid-intcmem  tty16    tty31  tty47  tty62  vcs3       vcsu5
gpiomem        media1        ram13      rpivid-vp9mem   tty17    tty32  tty48  tty63  vcs4       vcsu6

Anything else i can try? At a loss here. Any way to confirm if the usb port is dead?

Afterschnitzel, 2021-03-04:
Did you try to reflash MCU?

RcTomcat1, 2021-03-04:
> 
> 
> Did you try to reflash MCU?

How do i do that without specifying a USB Port? Make flash gives back "Please specify FLASH_Device".
I used make flash FLASH_DEVICE=MYSERIALDEVICE initially but that is no longer an option as there is no serial device recognized.

Afterschnitzel, 2021-03-04:
Try to flash it via CD card. I solved similar issue yesterday with another board and SD card flash helped me.

RcTomcat1, 2021-03-05:
@Bot thanks for closing....i wanted to determine whether or not it was a klipper issue or a hardware issue. If i would be certain that it was hardware related i would not have posted in the first place.....Hence the question: Any way to confirm if the usb port is dead?

By the way: Managed to get the board back up and running by placing a marlin firmware.bin on the sd card. Now it is running marlin again and the USB port is back to normal. So i still don't know if it is a) a hardware issue or b) somehow related to klipper.

Sineos, 2021-03-05:
Please follow the directions in https://www.klipper3d.org/Contact.html to get help

`/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0` usually indicates that the board is not correctly flashed with Klipper

closed 2021-03-06 by RcTomcat1