Klipper3d/klipper #4015: Accept CNC gcode variants

nytowl, 2021-03-03:
I'm trying to get klipper to work with an MPCNC and in general there aren't any problems but I've found most CNC design tools output a different syntax of gcode that klipper doesn't understand.

http://www.helmancnc.com/haas-cnc-mill-g-codes/

The basic issue is that the CNC output uses G00 instead of G0. Is the same for G0[1-9] vs G[1-9]

I think this is interpreted by the klipper firmware. Could it be modified to work with both types of gcode ?

As a work around could macros be used to map G00 to G0 ?

https://www.klipper3d.org/Config_Reference.html#g-code-macros-and-events

fleinze, 2021-03-05:
Can you test this macro?
```
[gcode_macro G01]
gcode:
   G1 {% if 'X' in params %}X{params.X} {% endif %}{% if 'Y' in params %}Y{params.Y} {% endif %}{% if 'Z' in params %}Z{params.Z} {% endif %}{% if 'E' in params %}E{params.E} {% endif %}{% if 'F' in params %}F{params.F}{% endif %}
```
Not sure if there is a more elegant solution.

mental405, 2021-03-06:
These should get the job done.

```
[gcode_macro G00]
gcode:  
    G1 {% for p in params %}{'%s%s' % (p, params[p])}{% endfor %}
    
[gcode_macro G01]
gcode: 
    G1 {% for p in params %}{'%s%s' % (p, params[p])}{% endfor %}

[gcode_macro G02]
gcode: 
    G2 {% for p in params %}{'%s%s' % (p, params[p])}{% endfor %}

[gcode_macro G03]
gcode: 
    G2 {% for p in params %}{'%s%s' % (p, params[p])}{% endfor %}

[gcode_macro G04]
gcode: 
    G4 {% for p in params %}{'%s%s' % (p, params[p])}{% endfor %}
```

closed 2021-04-04 by github-actions[bot]