Klipper3d/klipper #4016: SKR Mini V2.0 is not recognized

tobit6795, 2021-03-04:
So i flashed the SKR Mini v2.0 with Klipper build like this:

# STM32F103 with a "28KiB bootloader". Also, select "Enable extra
# low-level configuration options" and configure "GPIO pins to set at
# micro-controller startup" to "!PA14".

`usb 1-1.2: new full-speed USB device number 6 using xhci_hcd
usb 1-1.2: device descriptor read/64, error -32
usb 1-1.2: device descriptor read/64, error -32
usb 1-1.2: new full-speed USB device number 7 using xhci_hcd
usb 1-1.2: device descriptor read/64, error -32
usb 1-1.2: device descriptor read/64, error -32
usb 1-1-port2: attempt power cycle
usb 1-1.2: new full-speed USB device number 8 using xhci_hcd
usb 1-1.2: Device not responding to setup address.
usb 1-1.2: Device not responding to setup address.
usb 1-1.2: device not accepting address 8, error -71
usb 1-1.2: new full-speed USB device number 9 using xhci_hcd
usb 1-1.2: Device not responding to setup address.
usb 1-1.2: Device not responding to setup address.
usb 1-1.2: device not accepting address 9, error -71
usb 1-1-port2: unable to enumerate USB device

tobit6795, 2021-03-05:
Soo, my second SKR Mini V2.0 just arrived. I've completely rebuild the image, flashed again over SD Card.
Same issue here, so i think it's an firmware related issue

raymondh2, 2021-03-05:
Might be related to #3959 . This has been an issue I have seen a few times on the Voron Discord.

amikhalev, 2021-03-07:
FWIW, I am running commit 1950380d6c90e57f6fa844f886d208e5391ede7b on an SKR Mini V2.0 right now and it works.

Did you set the USB vendor/product IDs in the menuconfig?

tobit6795, 2021-03-11:
i retried everything on a clean system(freshly installed raspberry pi os) and with that image, everything worked. i dont now what caused this. I indeed cleared things up before every make (make clean / make distclean).

closed 2021-03-11 by tobit6795