Klipper3d/klipper #4019: ADXL345 installation fails with Error: No such file or directory: '/tmp/klipper_host_mcu'

l-Kage-l, 2021-03-04:
I just got an ADXL345 and i wanted to set it up for calibrating input shaping, I have and Ender 3 with a SKR E3 mini V1.2 and a Raspi4 with 4GB.
Steps i did:
I have updated to the latest klipper installation
Then i followed the steps here: https://www.klipper3d.org/Measuring_Resonances.html
Just after installing the additional dependencies for matlab, it says to follow the RPi Microcontroller document.
I went for that, installed the rc script just as described
Then i built the microcontroller code just as described
Then, for my board, i did a "make clean" then "make", pulled the klipper.bin out and flashed it to my board
After that, i went back to the Measuring Resonances site, added the additional commands to printer.cfg file, issued the RESTART command and now all i get is this:
Unable to open port: [Errno 2] No such file or directory: '/tmp/klipper_host_mcu'
Pulled from klippy.log
I searched for everything and couldnt find a solution.
And just as the error says, there is no file in the tmp folder name klipper_host_mcu.

What script/process was supposed to create the file? where am i mistaken?

l-Kage-l, 2021-03-04:
Never mind, got the solution from a previous post.

The solution:

After following the steps from the Rpi Microcontroller document, up to the "make menuconfig" option, you flash the new firmware.bin to the printer, then RE-run the "make menuconfig", select in the "boards" category the "linux process" then you type in "make clean", "make flash" and it will work. The last part, about re-running the script is not specified anywhere, thats why i was stuck for ages.

closed 2021-03-04 by l-Kage-l

jakep82, 2021-03-05:
You appear to have misread the document.  It makes no mention of flashing your printer board, and clearly says to select the Linux process.  From the document:

> To compile the Klipper micro-controller code, start by configuring it for the “Linux process”:

https://www.klipper3d.org/RPi_microcontroller.html#building-the-micro-controller-code

Pattern91, 2021-05-14:
Hi, i have the same problem...  I use an Orange Pi 2. I followed the instructions shown [here](https://www.klipper3d.org/RPi_microcontroller.html). But I don't get a file/link in /tmp/ which is called Klipper_host_mcu.

I don't know, what I am doing wrong. I selected the "Linux process" during the make menuconfig and also stopped Klipper before flashing. But nothing helps. The directory /usr/local/bin/klipper_mcu is created successfully.

My Klipper.log is attached below. I hope, you can help me...
[klippy.log](https://github.com/KevinOConnor/klipper/files/6479764/klippy.log)